verilator/verilator #3249: Build failing with `ld` error: file size too small (length = 8)

rachitnigam, 2021-12-17:
Verilator version: 4.216 2021-12-05 rev UNKNOWN.REV, Installed using `brew`
OS: MacOS 12.0.1 (21A559)
Files to reproduce: Run `build.sh` within the directory [repro.zip](https://github.com/verilator/verilator/files/7732174/repro.zip)

Expected: `verilator` command completes successfully
Actual:
```
clang++  -I.  -MMD -I/usr/local/Cellar/verilator/4.216/share/verilator/include -I/usr/local/Cellar/verilator/4.216/share/verilator/include/vltstd
-DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-opera
tion -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -
Wno-shadow      -std=gnu++14 -Os -c -o testbench.o ../testbench.cpp
clang++  -I.  -MMD -I/usr/local/Cellar/verilator/4.216/share/verilator/include -I/usr/local/Cellar/verilator/4.216/share/verilator/include/vltstd
-DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-opera
tion -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -
Wno-shadow      -std=gnu++14 -Os -c -o verilated.o /usr/local/Cellar/verilator/4.216/share/verilator/include/verilated.cpp
clang++  -I.  -MMD -I/usr/local/Cellar/verilator/4.216/share/verilator/include -I/usr/local/Cellar/verilator/4.216/share/verilator/include/vltstd
-DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-opera
tion -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -
Wno-shadow      -std=gnu++14 -Os -c -o verilated_vcd_c.o /usr/local/Cellar/verilator/4.216/share/verilator/include/verilated_vcd_c.cpp
/usr/bin/perl /usr/local/Cellar/verilator/4.216/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vmain.cpp Vmain___024root__DepSet_
hf7d72865__0.cpp Vmain___024root__DepSet_ha6e06459__0.cpp Vmain__Trace__0.cpp Vmain___024root__Slow.cpp Vmain___024root__DepSet_hf7d72865__0__Slow
.cpp Vmain___024root__DepSet_ha6e06459__0__Slow.cpp Vmain__Syms.cpp Vmain__Trace__0__Slow.cpp > Vmain__ALL.cpp
clang++  -I.  -MMD -I/usr/local/Cellar/verilator/4.216/share/verilator/include -I/usr/local/Cellar/verilator/4.216/share/verilator/include/vltstd
-DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-opera
tion -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -
Wno-shadow      -std=gnu++14 -Os -c -o Vmain__ALL.o Vmain__ALL.cpp
echo "" > Vmain__ALL.verilator_deplist.tmp
Archive ar -rcs Vmain__ALL.a Vmain__ALL.o
ar: @Vmain__ALL.a.verilator_deplist.tmp: No such file or directory
clang++    testbench.o verilated.o verilated_vcd_c.o Vmain__ALL.a      -o Vmain
ld: file too small (length=8) file 'Vmain__ALL.a' for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Vmain] Error 1
rm Vmain__ALL.verilator_deplist.tmp
%Error: make -C obj_dir -f Vmain.mk exited with 2
%Error: Command Failed /usr/local/Cellar/verilator/4.216/bin/verilator_bin -cc --trace out.sv --exe testbench.cpp --top-module main --build
```

Not sure if this is a Mac-specific issue. Previous stackoverflow answers seem to suggest Xcode might be to blame but I couldn't apply the [fix provided](https://stackoverflow.com/questions/14034048/xcode-linker-error-file-too-small-for-architecture-x86-64/40149836#40149836) because the folder in question doesn't exist.

wsnyder, 2021-12-17:
I suspect is a duplicate of #3031. Even if not close enough probably best to describe there, thanks.

closed 2021-12-17 by wsnyder

rachitnigam, 2021-12-22:
@wsnyder We were able to make things work by downgrading: `Verilator 4.200 2021-03-12 rev v4.110-43-g5022e81af (mod)` using an [older homebrew formula](https://github.com/Homebrew/homebrew-core/blob/082fe0d20516fd75c8d928e2e62d48d4db5ea1f8/Formula/verilator.rb). I can try to dig into things if it'd be helpful to figure out what changed between that version and latest.

(I wasn't sure if your last comment was meant to indicate that I should this discussion to #3031. Posting here because it resolved the specific error in this issue)

quark17, 2021-12-23:
I have the same experience: I am able to install 4.200 (5022e81af) via Homebrew on macOS (11.6.1); but a build with the head of the repo has those `file too small` errors.  So I am able to run with some version of Verilator, but I'd like to run with newer and also to be able to develop on the head of the repo.

If I check out 5022e81af from the repo, I can build and run that fine.  So it's not that the Homebrew recipe is doing something to make it work; I would guess that something has changed in Verilator since then?

In the other Verilator issue (3031), some people say they got it to work in some ways, but I didn't have luck replicating any of those.  One person said that they removed binutils from Homebrew and got it to work.  A websearch shows other projects where people have encountered the "too small" error and have hidden binutils/coreutils by removing the Homebrew directory from the path.  I tried building with `/usr/local/bin` (Homebrew's install location) removed from my `PATH`; but I still got the "too small" error when using it, whether I hid that path or not.  (Of course, if I remove the Homebrew path, then I don't have `autoconf` available on macOS; so for building, I'm calling it with an absolute path.)

The fact that I can build and run fine with 5022e81af but not with later versions seems interesting.  I could try finding the first commit where it stops working?

quark17, 2021-12-24:
The issue is introduced only 15 commits later in 6a189a1f, which edits `include/verilated.mk.in`.

Note that if you check out this commit, you also need to add `#include <array>` to `include/verilated_heavy.h`.  This was a bug introduced in ce79c4eb (which added code that needs array) but wasn't fixed until 0f7ec6c9b.

In any case, I don't quite understand what 6a189a1f has changed, to understand why it might have broken linking on macOS.

I notice that the CI jobs in this repo only test on Ubuntu.  GitHub also offers macOS environments; is there a reason not to add a macOS job?

wsnyder, 2021-12-24:
I'd welcome a macOs addition to the github action regressions, just need someone to provide the appropriate pull request, and critically be willing to signup to debug any failures that are non-obvious how to fix.

quark17, 2021-12-24:
Ah, the problem is that BSD `ar` does not support `@`, which is now used in `include/verilated.mk.in`.

Notice that before we get the error about "too small", we first get this warning:
```
ar: @Vmain__ALL.a.verilator_deplist.tmp: No such file or directory
```

FYI, the `make` on my macOS 11 is `GNU Make 3.81`, and it appears that 6a189a1 introduced something that only works for gmake 4, and there is a later commit that fixes this (44fd205e).  But the commit 6a189a1 is still where the problem is introduced, since it adds `@`, which is allowed by GNU `ar` to indicate a file containing a list of objects, in case they are too many to name on the command-line.

Is there some way to rewrite the code in `verilated.mk.in` so that it works on both GNU and BSD?  I notice that both support `-M` to provide an interactive script.  (I don't really know anything about `ar`, so I can't say if there's some way to use `xargs` here?)