verilator/verilator #3253: Policy clarification for verilator_ext_tests

gezalore, 2021-12-20:
Note: seems like issues are disabled for https://github.com/verilator/verilator_ext_tests so opening this one here.

The README there says:

> To be accepted in this package, an external submodule tested here must:
> 
> Be under an open source license.
> Be available as a git repository.
> Be self checking (potentially checked by the t/ script you write).
> Be Verilator lint clean (using lint_off pragmas in code is fine).
> Be willing to consider patches as needed (i.e. not a "dead project").
> Already run CI against a fixed version of Verilator

I think the verilator_ext_tests repo is very useful, and I think it could be even more helpful, but I have some issues with the above requirements

> Be available as a git repository.

This is kind of fair, but a git repo in and of itself is not always sufficient. I for example would like to add https://github.com/OpenXiangShan/XiangShan, but that project uses Chisel with a ton of dependencies that would be unfortunate to add to the test suite dependency. At the same time, one can just check in the generated Verilog and test Verilator against that. This project has found at least 2 bugs that did not show through the test suite.

Also, both XiangShan and OpenTitan require test executables for the simulated CPUs that are not part of the tree, and require the target toolchain, but are not too hard to conjure up once and commit.

> Be Verilator lint clean (using lint_off pragmas in code is fine).

I'm not sure why this is important, for example OpenTitan has tons of warnings, but is a very helpful test project. It's ok if it's a lint warning that indicates we will produce broken code or undefined behaviour, we don't want to be backward compatible with that, but things like width warnings that are not harmful or UNOPTFLAT should not cause a project to be rejected.

> Be willing to consider patches as needed (i.e. not a "dead project").

Somewhat of fair, but a big bag of Verilog that has a defined expected behaviour is still helpful for finding non-trivial issues.

> Already run CI against a fixed version of Verilator

Not sure why this is important

---

I would like to propose that we allow anything that is:
- Open source (so we can lawfully copy it's code if needed),
- and self checking,
- and either
  - Available as a git submodule without too many extraneous dependencies
  - or, can be committed into verilator_ext_tests or some other submodule under our control as a blob

We don't really *need to* ever update the bag of code type of tests. We just want to make sure the bag of code does the same thing before or after a Verilator change.

gezalore, 2021-12-20:
> The problem is there are times where the code does need to change ... the code does something which IEEE indicates is unpredictable, or new GCC tool chains give errors, or code had x86 assumptions and broke on Arm, 

FWIW, I was not suggesting to import garbage. These issues are rare in high quality projects, and if it is completely bust on ARM, then oh well, disable test on ARM. Still better than no testing.

As of having to change code because we added a warning sounds to me like pretty bad form from our side. If it's a genuine warning the user should be able to ignore it at their own peril, but this is besides the point of this issue.

The reason I brought this up is because I often find myself writing optimizations that should really only ever change the speed of the model, not anything else. For these the self checking blobs are perfectly fine.

> I'm not personally interested in spending a large effort to get someone else's project into Verilator and maintain it, especially as that probably means that project isn't a "customer"

Nor am I, I however have existence proofs where these large blobs found genuine issues of hard to hit corner cases, not because they are bad code, but because I typed in a bug, so I find them very useful in ensuring the "customers" we do care about will not run into the same problem 6 months down the line when I forgot about it all.

> If we add dead projects, then we'd need to understand a lot about the code when the inevitable portability issues arise when new toolchains come out.

Again, don't import garbage, but high quality code should not have many issues.

In any case what you propose above does not change the status quo so I won't be able to add these to this repo, maybe they don't belong there. I will continue to keep them elsewhere.

wsnyder, 2021-12-20:
I think we're agreed not to import garbage.  ;)

>As of having to change code because we added a warning sounds to me like pretty bad form from our side. If it's a genuine warning the user should be able to ignore it at their own peril, but this is besides the point of this issue.

Yes, typically a new warning results in only a suggestion or pull request upstream to either fix the code (because the warning was "right") or a one-liner to add a -Wno to their make scripts (because the warning was "noisy").  The much more common case is the extended case identifies a bug in the new Verilator warning which needs fixing.

>I won't be able to add these to this repo, maybe they don't belong there. I will continue to keep them elsewhere.

I of course agree there's value in more self-checking tests!

If you want to make an e.g. opentitan-on-verilator github project, then that opentitan-on-verilator project could then qualify for being a submodule in verilator_ext_tests.  You might also want to add opentitan-on-verilator to https://github.com/chipsalliance/sv-tests as parse-only so you get the scorecard versus more tools.

Having that extra project versus putting code under verilator itself perhaps seems silly, but I'd argue it's good as indicates to our community there's a division between the code which the verilator project as a whole is signed up to keep maintained, versus others are maintaining.

gezalore, 2021-12-20:
FWIW opentitan-on-verialtor is just opentitan :) but me making a fork just to add a few binaries and never ever update it again unless newer Verilator finds something is as you say, seems a bit odd but doable. I will proceed that way when I convinced myself I should do this.

closed 2021-12-20 by gezalore