verilator/verilator #3254: Compilation error on macOS Monterey with M1 Mac

adrienlemasle, 2021-12-21:
```
../V3Hash.h:59:40: error: ambiguous conversion for functional-style cast from 'unsigned long' to 'V3Hash'
        return V3Hash(combine(m_value, V3Hash(that).m_value));
                                       ^~~~~~~~~~~
../V3Hash.h:63:74: note: in instantiation of function template specialization 'V3Hash::operator+<unsigned long>' requested here
    template <class T> V3Hash& operator+=(T that) { return *this = *this + that; }
                                                                         ^
../V3Hasher.cpp:361:24: note: in instantiation of function template specialization 'V3Hash::operator+=<unsigned long>' requested here
                m_hash += itr.first;
                       ^
../V3Hash.h:37:14: note: candidate constructor
    explicit V3Hash(uint32_t val)
             ^
../V3Hash.h:39:14: note: candidate constructor
    explicit V3Hash(int32_t val)
             ^
../V3Hash.h:41:14: note: candidate constructor
    explicit V3Hash(uint64_t val)
             ^
../V3Hash.h:43:14: note: candidate constructor
    explicit V3Hash(int64_t val)
```

The compiler is:
```
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 13.0.0 (clang-1300.0.29.3)
Target: arm64-apple-darwin21.2.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

I think this is due to uint64_t being defined as unsigned long long for this compiler but vluint64_t being defined as unsigned long.

gezalore, 2021-12-21:
This seems to crop up again and again @wsnyder why do we need `vl*` flavorus? what is wrong with `uint*_t`/`int*_t` in `<cstdint>`?

wsnyder, 2021-12-21:
I was thinking the same thing.  These originally pre-dated there being the C++11 fixed types https://en.cppreference.com/w/cpp/types/integer

I'll try changing the typedefs to just reference the standard types.  If that doesn't prove a problem, after a few months we can replace all the code.

BTW there's t_dist_portability test that was supposed to detect using uint64_t instead of vluint64_t but that wasn't running against src for some reason.

wsnyder, 2021-12-21:
Try again please, should be fixed.

closed 2021-12-21 by wsnyder