verilator/verilator #3258: Verilator shows an internal error if a module that has inout ports is not inlined

yTakatsukasa, 2021-12-25:
If a module that has inout port is not inlined, verilator shows the following internal error.
```
%Error: Internal Error: t/t_tri_inout.v:8:32: ../V3Gate.cpp:1103: Consumer doesn't match lhs of assign
    8 |    io   io1(.A(A), .OE( SEL), .Z(Z), .Y(Y1));
      |                                ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
%Warning: vlt/t_tri_inout: Exec of perl failed: %Error: Internal Error: t/t_tri_inout.v:8:32: ../V3Gate.cpp:1103: Consumer doesn't match lhs of assign
```

This error can be easily reproduced by adding `/*verilator no_inline_module*/` metacomment to `io` module in `t_tri_inout.v`
https://github.com/verilator/verilator/blob/fd45be3def7e66b72684339ea455933472ff05db/test_regress/t/t_tri_inout.v#L18-L22

Passing `-Od` to skip dedupe optimization allows verilation, but the simulation result is not good.
I assume the optimization pass is essential.

The quickest fix would be forcing the module to be inlined regardless of its size, but not ideal of course.

I am trying to understand what dedupe optimization does and hopefully find better fix.
Any suggestion will be appreciated.

yTakatsukasa, 2021-12-27:
I don't fully understand tri-state handling yet, but output of V3Tristate looks strange.

```

wsnyder, 2021-12-29:
I suspect dedup is just catching the problem and not really related to the real issue with tristate.  It looks like the variable does have the wrong direction as you suggest - does fixing that make it work?

commit 4e5f3085 (2022-01-04): Fix #3258 of internal error with inout port (#3268)

* Tests: Modify t_tri_inout to reproduce #3258

* Set direction of __en accorting to its main signal direction

* Update Changes
closed 2022-01-04 by yTakatsukasa