verilator/verilator #3260: always_unroll pragma

FlyGoat, 2021-12-26:
What would you like added/supported?

When trying to verilate a huge existing project, I got tons of `BLKLOOPINIT `. Some of them can be resolved by rising `--unroll-count` and `--unroll-stmts`, but determine those values can be really painful. I'm trying to introduce always_unroll pragma for for loops so we can just supply it to for loops with dealyed assignment.

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)
No :-(

May we assist you in trying to fix this yourself?
Yes, but need some direction.

FlyGoat, 2021-12-26:
My failed attempt:
```
diff --git a/src/V3Ast.h b/src/V3Ast.h
index 14d58d1f..cdce1e15 100644
--- a/src/V3Ast.h
+++ b/src/V3Ast.h
@@ -228,7 +228,9 @@ public:
         PUBLIC_TASK,
         FULL_CASE,
         PARALLEL_CASE,
-        ENUM_SIZE
+        ENUM_SIZE,
+        ALWAYS_UNROLL,
+        NEVER_UNROLL
     };
     enum en m_e;
     inline AstPragmaType()
diff --git a/src/V3Unroll.cpp b/src/V3Unroll.cpp
index 22b14570..225b2afa 100644
--- a/src/V3Unroll.cpp
+++ b/src/V3Unroll.cpp
@@ -50,6 +50,8 @@ private:
     bool m_varModeReplace;  // Replacing varrefs
     bool m_varAssignHit;  // Assign var hit
     bool m_generate;  // Expand single generate For loop
+    bool m_alwaysUnroll;
+    bool m_neverUnroll;
     string m_beginName;  // What name to give begin iterations
     VDouble0 m_statLoops;  // Statistic tracking
     VDouble0 m_statIters;  // Statistic tracking
@@ -59,7 +61,7 @@ private:
 
     // VISITORS
     bool cantUnroll(AstNode* nodep, const char* reason) {
-        if (m_generate)
+        if (m_generate || m_alwaysUnroll)
             nodep->v3warn(E_UNSUPPORTED, "Unsupported: Can't unroll generate for; " << reason);
         UINFO(3, "   Can't Unroll: " << reason << " :" << nodep << endl);
         // if (debug() >= 9) nodep->dumpTree(cout, "-cant-");
@@ -91,6 +93,11 @@ private:
         AstNode* const precondsp, AstNode* condp,
         AstNode* const incp,  // Maybe under nodep or in bodysp
         AstNode* bodysp) {
+
+        if (m_neverUnroll) {
+            return cantUnroll(nodep, "never_unroll enabled");
+        }
+
         // To keep the IF levels low, we return as each test fails.
         UINFO(4, " FOR Check " << nodep << endl);
         if (initp) UINFO(6, "    Init " << initp << endl);
@@ -161,20 +168,22 @@ private:
             }
             if (!canSimulate(condp)) return cantUnroll(condp, "Unable to simulate condition");
 
-            // Check whether to we actually want to try and unroll.
-            int loops;
-            if (!countLoops(initAssp, condp, incp, unrollCount(), loops)) {
-                return cantUnroll(nodep, "Unable to simulate loop");
-            }
+            if (!m_alwaysUnroll) {
+                // Check whether to we actually want to try and unroll.
+                int loops;
+                if (!countLoops(initAssp, condp, incp, unrollCount(), loops)) {
+                    return cantUnroll(nodep, "Unable to simulate loop");
+                }
 
-            // Less than 10 statements in the body?
-            int bodySize = 0;
-            int bodyLimit = v3Global.opt.unrollStmts();
[... 86 lines cut ...]
 	;
 
 statementFor<beginp>:		// IEEE: part of statement
-		yFOR '(' for_initialization expr ';' for_stepE ')' stmtBlock
+		yFOR '(' for_initialization expr ';' for_stepE ')' loopVerilatorPragmas stmtBlock
 							{ $$ = new AstBegin($1, "", $3, false, true);
-							  $$->addStmtsp(new AstWhile($1, $4,$8,$6)); }
-	|	yFOR '(' for_initialization ';' for_stepE ')' stmtBlock
+							  if ($8) $$->addStmtsp($8);
+							  $$->addStmtsp(new AstWhile($1, $4,$9,$6));}
+	|	yFOR '(' for_initialization ';' for_stepE ')' loopVerilatorPragmas stmtBlock
 							{ $$ = new AstBegin($1, "", $3, false, true);
-							  $$->addStmtsp(new AstWhile($1, new AstConst($1,AstConst::BitTrue()), $7, $5)); }
+							  if ($7) $$->addStmtsp($7);
+							  $$->addStmtsp(new AstWhile($1, new AstConst($1,AstConst::BitTrue()), $8, $5));}
+	;
+
+loopVerilatorPragmas<nodep>:
+	/* nothing */       { $$ = nullptr; }
+	| 	yVL_ALWAYS_UNROLL	{ $$ = new AstPragma($1,AstPragmaType::ALWAYS_UNROLL); }
+	|	yVL_NEVER_UNROLL	{ $$ = new AstPragma($1,AstPragmaType::NEVER_UNROLL); }
 	;
 
 statementVerilatorPragmas<nodep>:
```

wsnyder, 2021-12-30:
Hmm, this seems a good attempt, and I agree having these pragmas seems a good idea. What doesn't work with this?  Can you please make it into a pull request with a test case for these?

I'd suggest unroll_always and unroll_never so they "sort" together in the documentation.

wsnyder, 2022-06-04:
Do you have interest in finishing off these new pragmas, or are you all set without?

FlyGoat, 2022-06-04:
> Do you have interest in finishing off these new pragmas, or are you all set without?

Ah I just got a little bit lost. As I don’t know how to debug when it not work…

wsnyder, 2022-06-05:
If you run with "--debug --debugi-V3Unroll 9" you'll see the UINFO statements print out and you can add new ones to see what's going on.  Also the renames and test can be added.  If you get stuck I can take it further.