verilator/verilator #3261: How do I use Verilator on Windows 10 ?

tushar-uwaterloo, 2021-12-28:
I used Windows 10 WSL to install Verilator on my machine.

Now, I have an application that needs to run verilator using Powershell. And so, 
(1) I installed `perl` on windows
(2) I mapped the Verilator directory (`\\wsl$\Ubuntu\`) to a new drive `P:`
(3) I can invoke Verilator now, using powershell command `>> perl P:/verilator/bin/verilator`

The problem comes when I try to actually do something meaningful with Verilator, like run lint check on a file. And so, I run on Powershell `>> perl P:/verilator/bin/verilator --lint-only -Wall myfile.sv`

here I get the following error:
`'verilator_bin' is not recognized as an internal or external command,
operable program or batch file.`

Digging further, I realized that `bin/verilator` file do a system call to `bin/verilator_bin` file, which windows Powershell does not understand. Is there a way I can still use Verilator on windows?

wsnyder, 2021-12-29:
I don't personally use WSL much so perhaps others can better advise.  But I'd recommend running verilator instead under the WSL distribution's shell instead of powershell.

wsnyder, 2022-01-01:
Closing as answered, but feel free to follow up here.

closed 2022-01-01 by wsnyder