verilator/verilator #3264: Internals: Improve tree syntax checks

wsnyder, 2022-01-02:
@gezalore suggested some stronger node documentation which is probably best implemented as enforcement in e.g. V3Broken:

- [ ] What nodes are legal in between each processing stage.   Perhaps for each node we have an "is it legal to have it" boolean.  A visitor stage can then clear that flag on the nodes that are no longer valid.  V3Broken will complain if it exists.
- [ ] What op1/2/3/4p's can have lists versus non-lists.  This is currently captured somewhat between use of addNOp versus setNOp, but that is not checked in V3Broken, and edits to the tree can get around those function's checks.
- [ ] What node types are allowed as each child of a node.
- [X] Checking for width consistency, see #3245.

Additional ideas welcome.  I'm going to start on some of these...

gezalore, 2022-01-02:
> What nodes are legal in between each processing stage. Perhaps for each node we have an "is it legal to have it" boolean. A visitor stage can then clear that flag on the nodes that are no longer valid. V3Broken will complain if it exists.

It would be best not to have this in the node itself. Instead we could have a static array of bools indexed by node type. Each pass can then set or clear which types are valid. V3Broken can then just `UASSERT(node_type_allowed[nodep->type()])`. This is complicated a bit by the conditionals in `Verilator.cpp` `process()`, which disable some passes in some cases. A fix for that would be to simply always apply a pass and in the pass skipp processing when unnecessary.

> What op1/2/3/4p's can have lists versus non-lists. This is currently captured somewhat between use of addNOp versus setNOp, but that is not checked in V3Broken, and edits to the tree can get around those function's checks.

I would say this would be nice to have for error checks, but node semantics usually define what makes sense, so for clarity I would say this is low priority.

> What node types are allowed as each child of a node.

My preferred way of doing this is to strengthen the type signatures of setters/getters. Maybe the solution for both this and the previous is to rely on declarative metacomments processed by `astgen`:

```
class AstFoo : public AstNode {
// op1 is bar: AstNode
// op2 is baz: List[AstNode]
ASTGEN_ACCESSORS_Foo; // Macro generated by astgen (similar to ASTGEN_SUPER_Foo) that declares bar/baz accessors. astgen can also then generate assertions to be used in V3Broken)
}
```

wsnyder, 2022-01-02:
For legal nodes, I'm prototyping code that looks like this in e.g. V3Assert.cpp bottom:

```
void V3Assert::assertAll(AstNetlist* nodep) {
    UINFO(2, __FUNCTION__ << ": " << endl);
    { AssertVisitor{nodep}; }  // Destruct before checking
    V3Broken::vnTypeAllowed(VNType::atAssert, false);
    V3Broken::vnTypeAllowed(VNType::atAssertIntrinsic, false);
    V3Broken::vnTypeAllowed(VNType::atCover, false);
    V3Broken::vnTypeAllowed(VNType::atMonitorOff, false);
    V3Broken::vnTypeAllowed(VNType::atPast, false);
    V3Broken::vnTypeAllowed(VNType::atRestrict, false);
    V3Broken::vnTypeAllowed(VNType::atSampled, false);
    V3Global::dumpCheckGlobalTree("assert", 0, v3Global.opt.dumpTreeLevel(__FILE__) >= 3);
}
```

gezalore, 2022-04-27:
> For legal nodes, I'm prototyping code that looks like this in e.g. V3Assert.cpp bottom:
>
> ```
> void V3Assert::assertAll(AstNetlist* nodep) {
>     UINFO(2, __FUNCTION__ << ": " << endl);
>     { [...]

Wonder what state this is in, I'm more and more finding myself needing my assumptions validated regarding the structure of the  Ast at various stages, so might have a go at implementing this myself.

wsnyder, 2022-04-27:
Due to some optimizations etc getting skipped for XML and lint only, etc, this ended up being a mess and I haven't followed up, if you are interested I can send the patch with the mechanics similar to described above.

I suspect the 4th bullet (What node types are allowed as each child of a node) is most valuable and I haven't done anything on that. Would you like to make a proposal?  I think the checks need to be done in V3Broken, the getters/setters are not sufficient as many tree edits are done by grafting and that would escape such checks.

gezalore, 2022-04-27:
> Due to some optimizations etc getting skipped for XML and lint only, etc, this ended up being a mess and I haven't followed up, if you are interested I can send the patch with the mechanics similar to described above.

I think this would still be very useful if the checks were only enforced if we are doing proper compilation.

> I suspect the 4th bullet (What node types are allowed as each child of a node) is most valuable and I haven't done anything on that. Would you like to make a proposal?

Actually, I think here we need some pretty significant refactoring for this to be valuable, as some of the nodes are quite ad-hoc. E.g.: in an ideal world, a compound AstNodeStmt should only contain sub AstNodeStmts, (e.g.: if-branches, etc) and AstNodeMath should only contain AstNodeMath children, but this breaks down in the current world pretty quickly due to things like calls, that can be either. These can be solved by  "splice" nodes, so a call would be an AstNodeMath, and there would be an AstStmtMath, which is the "expression in statement position node". All this would be good in the long run to make visitors clearer, but I'm a bit wary of the effort right now (though would like to do this one day). For some inspiration I recommend taking a peak at the AST ADTs of Alogic, which have worked pretty well for me before:
https://github.com/alogic/alogic/blob/6f3a0b958f4b7d18d1d40ddafb773e1983bf1c43/src/main/scala/com/argondesign/alogic/ast/Trees.scala#L260-L289

> I think the checks need to be done in V3Broken, the getters/setters are not sufficient as many tree edits are done by grafting and that would escape such checks.

I think getters and setters are enough go auto-generate the rest of the checks that are then enforced in V3Broken (or even during editing, in a debug build).

wsnyder, 2022-05-02:
Attached is the patch I started on this - intended mostly for conceptual reference as I didn't really love how it was heading.

[bug3264_legal_at_stage.patch.txt](https://github.com/verilator/verilator/files/8600482/bug3264_legal_at_stage.patch.txt)

wsnyder, 2024-10-04:
I think the last item here, what-nodes-in-what stage, simply got too ugly and was abandoned over a year ago.  Other item are implemented earlier. I'm going to close this out as it's unlikely to get more attention.

closed 2024-10-04 by wsnyder