verilator/verilator #3266: Makefile changes in `ar` (version 4.204) broke building on Mac

carlosedp, 2022-01-03:
Changes in `ar` in version 4.204 broke building on Mac using Homebrew.

Previously, on version 4.200 (one before last in Homebrew), my projects build successfully:

```
❯ make verilator
verilator -O3 --assert --cc generated/GPIOInOut.v --cc generated/PLL.v --cc generated/Toplevel.v --exe verilator/chiselv.cpp verilator/uart.c --top-module Toplevel -o chiselv --timescale 1ns/1ps
make -C obj_dir -f VToplevel.mk -j`nproc`
make[1]: Entering directory '/Users/cdepaula/repos/chiselv/obj_dir'
g++  -I.  -MMD -I/Users/cdepaula/verilator-install/share/verilator/include -I/Users/cdepaula/verilator-install/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o chiselv.o ../verilator/chiselv.cpp
g++  -I.  -MMD -I/Users/cdepaula/verilator-install/share/verilator/include -I/Users/cdepaula/verilator-install/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o uart.o ../verilator/uart.c
g++  -I.  -MMD -I/Users/cdepaula/verilator-install/share/verilator/include -I/Users/cdepaula/verilator-install/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o verilated.o /Users/cdepaula/verilator-install/share/verilator/include/verilated.cpp
/usr/bin/perl /Users/cdepaula/verilator-install/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VToplevel.cpp VToplevel__Slow.cpp VToplevel__Syms.cpp > VToplevel__ALL.cpp
g++  -I.  -MMD -I/Users/cdepaula/verilator-install/share/verilator/include -I/Users/cdepaula/verilator-install/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o VToplevel__ALL.o VToplevel__ALL.cpp
clang: warning: treating 'c' input as 'c++' when in C++ mode, this behavior is deprecated [-Wdeprecated]
Archive ar -cr VToplevel__ALL.a VToplevel__ALL.o
g++    chiselv.o uart.o verilated.o VToplevel__ALL.a      -o chiselv
make[1]: Leaving directory '/Users/cdepaula/repos/chiselv/obj_dir'
```

But recent Homebrew update to 4.216, I started getting the error:

```
❯ make verilator
verilator -O3 --assert --cc generated/GPIOInOut.v --cc generated/PLL.v --cc generated/Toplevel.v --exe verilator/chiselv.cpp verilator/uart.c --top-module Toplevel -o chiselv --timescale 1ns/1ps
make -C obj_dir -f VToplevel.mk -j`nproc`
make[1]: Entering directory '/Users/cdepaula/repos/chiselv/obj_dir'
g++  -I.  -MMD -I/Users/cdepaula/verilator-install/share/verilator/include -I/Users/cdepaula/verilator-install/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=c++14 -Os -c -o chiselv.o ../verilator/chiselv.cpp
g++  -I.  -MMD -I/Users/cdepaula/verilator-install/share/verilator/include -I/Users/cdepaula/verilator-install/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=c++14 -Os -c -o uart.o ../verilator/uart.c
g++  -I.  -MMD -I/Users/cdepaula/verilator-install/share/verilator/include -I/Users/cdepaula/verilator-install/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=c++14 -Os -c -o verilated.o /Users/cdepaula/verilator-install/share/verilator/include/verilated.cpp
/usr/bin/perl /Users/cdepaula/verilator-install/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VToplevel.cpp VToplevel___024root__DepSet_hf000d6bc__0.cpp VToplevel___024root__Slow.cpp VToplevel___024root__DepSet_hf000d6bc__0__Slow.cpp VToplevel__Syms.cpp > VToplevel__ALL.cpp
echo "" > VToplevel__ALL.verilator_deplist.tmp
g++  -I.  -MMD -I/Users/cdepaula/verilator-install/share/verilator/include -I/Users/cdepaula/verilator-install/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=c++14 -Os -c -o VToplevel__ALL.o VToplevel__ALL.cpp
clang: warning: treating 'c' input as 'c++' when in C++ mode, this behavior is deprecated [-Wdeprecated]
Archive ar -rcs VToplevel__ALL.a VToplevel__ALL.o
ar: @VToplevel__ALL.a.verilator_deplist.tmp: No such file or directory
g++    chiselv.o uart.o verilated.o VToplevel__ALL.a      -o chiselv
ld: file too small (length=8) file 'VToplevel__ALL.a' for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [VToplevel.mk:68: chiselv] Error 1
rm VToplevel__ALL.verilator_deplist.tmp
make[1]: Leaving directory '/Users/cdepaula/repos/chiselv/obj_dir'
gmake: *** [Makefile:63: verilator] Error 2
```

and tracked down to the changes in https://github.com/verilator/verilator/blob/e3341e9a7ce09592e498b23bf759ef16b09cbcc1/include/verilated.mk.in#L206-L234

carlosedp, 2022-01-03:
My version is:

```
❯ verilator --version
Verilator 4.216 2021-12-05 rev UNKNOWN.REV
```

Here's my project dir with Verilator output files:

[obj_dir.zip](https://github.com/verilator/verilator/files/7802331/obj_dir.zip)

wsnyder, 2022-01-03:
See #3256

carlosedp, 2022-01-03:
Thanks! I can confirm that with #3256, building on Mac works:

```
❯ verilator --version
Verilator 4.217 devel rev v4.216-70-g9dc462ea

❯ make verilator
verilator -O3 --assert --cc generated/GPIOInOut.v --cc generated/PLL.v --cc generated/Toplevel.v --exe verilator/chiselv.cpp verilator/uart.c --top-module Toplevel -o chiselv --timescale 1ns/1ps
make -C obj_dir -f VToplevel.mk -j`nproc`
make[1]: Entering directory '/Users/cdepaula/repos/chiselv/obj_dir'
g++  -I.  -MMD -I/Users/cdepaula/verilator-install/share/verilator/include -I/Users/cdepaula/verilator-install/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o chiselv.o ../verilator/chiselv.cpp
g++  -I.  -MMD -I/Users/cdepaula/verilator-install/share/verilator/include -I/Users/cdepaula/verilator-install/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o uart.o ../verilator/uart.c
g++  -I.  -MMD -I/Users/cdepaula/verilator-install/share/verilator/include -I/Users/cdepaula/verilator-install/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o verilated.o /Users/cdepaula/verilator-install/share/verilator/include/verilated.cpp
/usr/bin/perl /Users/cdepaula/verilator-install/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VToplevel.cpp VToplevel___024root__DepSet_hf000d6bc__0.cpp VToplevel___024root__Slow.cpp VToplevel___024root__DepSet_hf000d6bc__0__Slow.cpp VToplevel__Syms.cpp > VToplevel__ALL.cpp
echo "" > VToplevel__ALL.verilator_deplist.tmp
g++  -I.  -MMD -I/Users/cdepaula/verilator-install/share/verilator/include -I/Users/cdepaula/verilator-install/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o VToplevel__ALL.o VToplevel__ALL.cpp
clang: warning: treating 'c' input as 'c++' when in C++ mode, this behavior is deprecated [-Wdeprecated]
Archive ar -rcs VToplevel__ALL.a VToplevel__ALL.o
g++    chiselv.o uart.o verilated.o VToplevel__ALL.a      -o chiselv
rm VToplevel__ALL.verilator_deplist.tmp
make[1]: Leaving directory '/Users/cdepaula/repos/chiselv/obj_dir'
```

closed 2022-02-10 by wsnyder