verilator/verilator #3270: Verilator forces all members of an interface to have either blocked or non-blocking assignments

jotego, 2022-01-05:
Verilator linter marks it as an error when a register is used in both blocked and non-blocking assignments. That is fine. But, when a System Verilog interface is comprised of many logic elements, I think it is legit to have some of them use blocked assignments while others use non-blocking.

For instance, when parsing the files in the [fx68k project](https://github.com/ijor/fx68k), there is an interface called *s_nanod*:
```
typedef struct {
	// ... other definitions
	logic updSsw;
	logic ryl2ab;
} s_nanod;
```
And Verilator complaints because of this:
```
%Error-BLKANDNBLK: ../../../../modules/fx68k/fx68k.sv:293:10: Unsupported: Blocked and non-blocking assignments to same variable: 'jtcop_game.u_main.u_cpu.u_cpu.Nanod'
  293 |  s_nanod Nanod;
      |          ^~~~~
                   ../../../../modules/fx68k/fx68k.sv:790:9: ... Location of blocking assignment
  790 |  assign Nanod.updSsw = Nanod.aob2Ab;
      |         ^~~~~
                   ../../../../modules/fx68k/fx68k.sv:917:4: ... Location of nonblocking assignment
  917 |    Nanod.ryl2ab <= Nanod.reg2abl & !ablSpecial & nanoLatch[ NANO_RXL_DBL];
      |    ^~~~~
```
The error applies to the interface variable _Nanod_, but it arises from individual elements using different assignments. Each one of them only uses one type of assignment.

I believe this use is correct even though Verilator marks this as an error.

wsnyder, 2022-01-07:
Verilator currently converts all unpacked structs to packed structs (#1541).  Thus once this is done, the register is effectively used both ways, as the lint rules are whole signal, not bit-by-bit as that corresponds to how the ordering is done.

One workaround might be to use /*verilator split_var*/

wsnyder, 2022-01-09:
Closing as is due to packed structs, feel free to follow up here though.

closed 2022-01-09 by wsnyder