verilator/verilator #3271: Inconsistent behavior when providing `-Mdir` versus relying on the default `obj_dir`

GeorgeLyon, 2022-01-07:
When I call Verilator without the `-Mdir` argument, I get a directory named `obj_dir` with all of the output _and_ a `vlt` directory which roughly has the contents of `$VERILATOR_ROOT` (i.e. a `bin` and an `include` folder with files like `verilated.mk`). When I provide the `-Mdir` argument, the artifacts are written to the specified folder, but no `vlt` directory is created.

1. Is this expected behavior? I don't see this documented anywhere in the documentation.
2. Is there a way to _force_ the emission of the `vlt` directory, even when using `-Mdir`? (this makes the output more portable than referring to an absolute path on the file system)

Interestingly, this behavior happens for me on when I use `Verilator 4.211 devel rev v4.210-59-g3ec3c2c2` (docker's `verilator/verilator` image) but not `Verilator 4.028 2020-02-06 rev v4.026-92-g890cecc1` (from `apt-get`).

```
📁 ~/Developer/Projects/Edith/Local Packages/Verilator/Tests/VerilatorTests/Adder
🔸 verilator --cc --exe -CFLAGS "-fPIC -shared -fvisibility=hidden" -LDFLAGS "-shared -dynamiclib -fvisibility=hidden" Adder.sv --top-module Adder CVerilatorSupport.cpp  

📁 ~/Developer/Projects/Edith/Local Packages/Verilator/Tests/VerilatorTests/Adder
🔸 ls obj_dir  
VAdder.cpp					VAdder___024root__DepSet_fe519e27__0__Slow.cpp
VAdder.h					VAdder___024root__Slow.cpp
VAdder.mk					VAdder__ver.d
VAdder__Syms.cpp				VAdder__verFiles.dat
VAdder__Syms.h					VAdder_classes.mk
VAdder___024root.h				vlt
VAdder___024root__DepSet_fe519e27__0.cpp

📁 ~/Developer/Projects/Edith/Local Packages/Verilator/Tests/VerilatorTests/Adder
🔸 verilator --cc --exe -CFLAGS "-fPIC -shared -fvisibility=hidden" -LDFLAGS "-shared -dynamiclib -fvisibility=hidden" Adder.sv --top-module Adder CVerilatorSupport.cpp -Mdir verilated

📁 ~/Developer/Projects/Edith/Local Packages/Verilator/Tests/VerilatorTests/Adder
🔸 ls verilated 
VAdder.cpp					VAdder___024root__DepSet_fe519e27__0.cpp
VAdder.h					VAdder___024root__DepSet_fe519e27__0__Slow.cpp
VAdder.mk					VAdder___024root__Slow.cpp
VAdder__Syms.cpp				VAdder__ver.d
VAdder__Syms.h					VAdder__verFiles.dat
VAdder___024root.h				VAdder_classes.mk
```

wsnyder, 2022-01-07:
Verilator itself never copies files to a "vlt" directory.  However, the docker script (see ci/docker/run/verilator-wrap.sh) does make this directory when a build finishes.   So that explains the difference.

Note bug #2358 which discusses an attempt to clean that up.  We'd welcome input or a patch to improve that.

Suggest we discuss it in that thread so closing here.

closed 2022-01-07 by wsnyder

GeorgeLyon, 2022-01-07:
Thanks for clarifying!