verilator/verilator #3275: Support interface in the top module when creating a library (--lib-create, --protect-lib, and --hierarchical)

yTakatsukasa, 2022-01-08:
The top module is not allowed to have an interface port.

Unlike input/output ports, interface port needs to be tied with interface cell.
V3LinkDot has considerable code for interface handling.

I am thinking of the following approach though I'm still reading the related code 
- modify interface port to interface cell in the top module in early stage of verilation
- treat the interface cell as an interface port when emitting SystemVerilog wrapper

wsnyder, 2022-01-08:
I think that's a reasonable approach but a complication to test is if the interface both crosses the boundary and is used internally.  You'll need to rename the interface to the cell with some suffix, and keep the old interface for internal use.

BTW sorry interface code is such a mess, it never seemed to fit well into the concepts, perhaps I missed some abstraction or something to simplify the concepts.

wsnyder, 2022-01-08:
Not sure of ready vs assigned - don't know if you meant were starting on this. Anyhow we don't really sweat active vs assigned right now.