verilator/verilator #3277: Duplicate declaration of CELLINLINE 'genblk1': 'genblk1'

dpetrisko, 2022-01-10:
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

Hi, I'm having an issue that suddenly appeared with minor changes in my codebase. When I instantiate the following code in a for loop in a large design, I get the following error: 

https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_cache/bsg_cache.v

```
%Error: /mnt/users/ssd0/no_backup/petrisko/pre-alpha-release/external/basejump_stl/bsg_cache/bsg_cache.v:179:51: Duplicate declaration of CELLINLINE 'genblk1': 'genblk1'
                                                                                                               : ... In instance testbench.wrapper.unicore.dut.c[0].dut.l2s.bank[0].cache
  179 |   else if (burst_len_lp == block_size_in_words_p) begin
      |                                                   ^~~~~
        /mnt/users/ssd0/no_backup/petrisko/pre-alpha-release/external/basejump_stl/bsg_cache/bsg_cache.v:123:51: ... Location of original declaration
  123 |   else if (burst_len_lp == block_size_in_words_p) begin
      |                                                   ^~~~~
%Error: /mnt/users/ssd0/no_backup/petrisko/pre-alpha-release/external/basejump_stl/bsg_cache/bsg_cache.v:571:51: Duplicate declaration of CELLINLINE 'genblk1': 'genblk1'
                                                                                                               : ... In instance testbench.wrapper.unicore.dut.c[0].dut.l2s.bank[0].cache
  571 |   else if (burst_len_lp == block_size_in_words_p) begin
      |                                                   ^~~~~
        /mnt/users/ssd0/no_backup/petrisko/pre-alpha-release/external/basejump_stl/bsg_cache/bsg_cache.v:179:51: ... Location of original declaration
  179 |   else if (burst_len_lp == block_size_in_words_p) begin
      |                                                   ^~~~~
```

I tried to shrink the test case, but anything smaller than my full project doesn't appear to trigger the inlining in an erroneous way. I can confirm that adding labels to these genblks gets rid of the issue. Ideally, I would like to solve this without modifying the source RTL though.

What 'verilator --version' are you using?  Did you try it with the git master version?

Verilator 4.213 devel rev v4.212-46-g20af8a36

What OS and distribution are you using?

CentOS Linux release 7.9.2009

May we assist you in trying to fix this in Verilator yourself?

More interested in a workaround. I'm happy to send full logs with my build if that would be helpful. Thanks!

gezalore, 2022-01-10:
This is probably due to my recent change to the I lining pass. I will have a look, but exact reproduction instructions would be helpful.

gezalore, 2022-01-10:
Hmm actually I think I misread and you don't have that patch in so not sure what is going on.

wsnyder, 2022-01-10:
It is worth trying -Oi, this might make it show up with smaller files.

dpetrisko, 2022-01-16:
This appears to be fixed in Verilator 4.217 devel rev v4.216-88-gf8c0169e. Feel free to close unless we want to coerce this example into a regression test

closed 2022-01-16 by wsnyder