verilator/verilator #3278: Redesign of scheduling (fixing generated clocks)

gezalore, 2022-01-15:
This issue is for the discussion and coordination of a proposed redesign of the
implementation of scheduling of internally derived events (often referred to as 
generated clocks) in Verilator.

# Background

The execution semantics of a Verilator generated model with clock signals only
appearing as top level inputs is clear and well defined. There is however some
uncertainty about the behaviour when clocks are generated inside the verilated
design itself. While the current implementation often works well, there can be
issues that cause difficulty by either resulting in incorrect simulation, or by
differing from other simulators. There are also use cases where the designer's
intent is to model an external clock generator with a specific behaviour, which
while not necessary well defined according to IEEE 1800, is nevertheless a
useful use case to support.

See for example #2557, #2653, #2807, #3047, #3175, #3178, #3269.

# Goals

The goal of the work proposed here is to come up with a scheme that both allows
reasoning about the simulation of derived events, and lends itself to high
performance simulation. This in turn should enable us to:
- Describe the execution semantics of a design precisely, so the expected
  behaviour of a simulation can be predicted.
- Be compliant with the IEEE 1800 scheduling semantics.
- Adjust execution semantics as desired, to either match other simulators, or
  to support modeling of advanced clocking schemes.

Additionally it would be beneficial if the scheme did not hinder possible
future implementation of generic dynamic event scheduling.

# Terminology

Before going into details, I would like to propose some terminology to make
the conversation clear.

I would like to refer to the current state of any variable that is driven via a
non-blocking assignment, as the `_q` of the variable, while the new value
computed on the right hand side of the non-blocking assignment as the `_d` of
the variable, analogous to the common labeling of the corresponding pins of a D
flip-flop. E.g.:

```verilog
reg x;

always @(posedge clk) x <= y ;
```

In the above, reading `x` yields `x_q`, while in this example `x_d` can be
considered always equal to `y`. This example, could also be rewritten in the
sometimes preferred style of pulling all computation into combinational blocks:

```verilog
reg x_q, x_d;

always_comb x_d = y;

always @(posedge clk) x_q <= x_d;
```

Note, the above rewrite is valid if and only if the computation of the `x_d` is
pure (has no side effect). Also note that the concept of `x_d` is more subtle
than "the right hand side of the non-blocking assignment:

```verilog
always @(posedge clk) begin
  if (en) x <= y;
end
```
[... 373 lines cut ...]
- To be standards compliant, would we need to trigger events on transient
  changes?  (i.e.: if a posedge only appears during an iteration of the settle
  loop, but not once the circuit is settled). This is currently not considered
  and would not work.

# Next steps

To be able to answer the questions about correctness above, we need to gather
examples that are small (so they can be analyzed by a human) but demonstrate
the scheduling problems. Think of this as a conformance test suite.

To appreciate the complexity of the problem, it would be useful to write a
Verilator pass that analyses the scheduling domains (more generic than a clock
domain) of the input design. This should output a graph of non-coincident
events and edges denoting which event can trigger which other event in a
combinational (Active region -> Active region), or a sequential (NBA region ->
Active region) manner.

# Further considerations

It would also be very useful if we could correctly simulate separately
verilated models together with arbitrary interfaces between. Think of this as
an improved hierarchical mode. I think this might be possible by chopping the
eval loops into parts and connecting them together in the desired way to form a
single effective loop.

gezalore, 2022-01-15:
@wsnyder  Summary so far + more background and context for other readers. More ideas to come but if any of the above is known to be incorrect (esp in the "Update Algorithm" section), I would appreciate if we caught it before we go further. Apologies for the verbosity.

gezalore, 2022-01-16:
Edit: Reformat

gezalore, 2022-01-16:
To support user annotated coincident events (#3178), I believe we need to support what is the moral equivalent of blocking assignments in clocked processes:

```verilog
reg clk
reg clk_half;

always @(posedge clk) begin
   clk_half = ~clk_half;
end
```

Once we can do this (and schedule it correctly), one can just rewrite signals generated via blocking assignments as non-blocking, with some further careful substitutions of `_q` with `_d` where necessary so the correct values are picked up.

gezalore, 2022-01-16:
Here is an test that should pass, but currently does not due to scheduling:

https://github.com/gezalore/verilator/blob/bc11f80baed02b63bb00898897dd1eef336a9da0/test_regress/t/t_scheduling_0.v

wsnyder, 2022-01-22:
# Terminology/Scheduling

>Combinational logic is not considered part of any "event domain" but can be pushed into one as an optimization under the right circumstances.

To avoid future confusion, I think that statement mixes two concepts. It is reasonable to have (but doesn't Verilate correctly):

```
   always @ (posedge clk or event_signal)
       active = d;  // Schedules to active
       inactive_q <= #0 d;   // Schedules to inactive region
       nba_q <= d;  // Schedules to NBA
```

That is there's a given event expression, which activates a process, and that process may activate variables in different scheduling region(s).

Crazy things like this have been spotted in PLL and memory models.

# Update Algorithm

It might be clearer to reword algorithm section to refer more to the IEEE region names.

This assumes (as you later note) that there are no combinatorial events.  Perhaps we should consider something where that's supported (more below).

>Current Verilator [does not implement] more generic cases of mutually exclusive subsets of event domains, for example posedge vs negedge triggered blocks on the same clock that can never execute together during a single evaluation.

Note I've seen one design where using --threads on a nearly all posedge clocked design was using about 10% of the time just to do the negedge eval(). This was because it still had all of the mtasks, and had to send the appropriate semaphones through all those, where in reality they could have all been skipped with a simple if statement in the top eval.

Also note I've tried experiments where resets were pulled out completely and the code duplicated under the theory that such reset code never executes, and all the "if (!reset)" and similar logic can then get removed on the fastpath.  This had only insiginficant performance improvement.

# Implementation Ideas

>A plausible way to implement this with minimal changes to Verilator is to turn clock sensitivities into explicit one-shot events.

I wonder if we should also be making events for processes with combinatorial logic?

Also when we have time-triggered events this will allow them to trigger such processes. 

e.g. the loop is more like:

```c
update_state_desired_version {

  do {
    Compute active triggered event domains, which includes "combinatorial events"

    Evaluate all actives within triggered processes
  } while (!settled);
```

Or perhaps that is a separate later change.

Assuming we add more internally generated events for clocked events or otherwise, there's probably additional items to add to your "optimizations" list.

>  this->compute_one_shot_trigger_events_dependent_on_inputs();
>Note the above is an idea that might be sufficient an requires relatively small modification to the existing code.

I like this concept as a good incremental change, which will also help us learn a great deal.

# Questions

>- Is there a use case for a non-blocking assignment in a combinational block?  This is currently not considered and would not work in the general case.

As above, there are cases, but not anything Verilator supports now.

>- To be standards compliant, would we need to trigger events on transient changes?  (i.e.: if a posedge only appears during an iteration of the settle loop, but not once the circuit is settled). This is currently not considered and would not work.

I think technically IEEE requires triggering on transients, but realistically this is unlikely to be needed, and I suspect other hybrid simulators likewise skip doing that.

# Further considerations

>improved hierarchical mode. I think this might be possible by chopping the eval loops

To give others background the basic idea here is

```verilog
  input a, b;
  output ao; assign ao = a;
  output bo; assign bo = b;
```

Should become two separate eval loops, one sensitive to a, and one to b.  This should likewise propagate to two different sensitivity processes (for a and b) in SystemC.

```
    SC_METHOD(eval_a);
    sensitive << a;

    SC_METHOD(eval_b);
    sensitive << b;
```

If instead of SC using --hierarchical, the upper hierarchical model would call eval_a and eval_b as appropriate.  Most importantly if the code in the upper level has

    assign b = a0;

This allows the upper eval to do ordering (e.g call 1. eval_a, 2. assign b=a0, 3. eval_b) without any change detects.

gezalore, 2022-05-15:
This is now implemented on the [develop-v5](/verilator/verilator/tree/develop-v5) branch and is ready for wider testing. Please report any feedback and problems under new issues.

closed 2022-05-15 by gezalore