verilator/verilator #3280: Eliminate assignments to dead variables

gezalore, 2022-01-16:
Verilating:

```verilog
  always @(posedge clk) a <= b;
  always @(posedge clk) b <= a;
```

yields:

```
  __Vdly__a = vlSelf->a;
  __Vdly__a = vlSelf->b;
  vlSelf->b = vlSelf->a;
  vlSelf->a = __Vdly__a;
```

The first assignment is redundant. I wonder how many of these are there that the C++ compiler can't get rid of due to function splitting.

wsnyder, 2022-01-16:
V3Life should have gotten rid of the duplication.  I believe it doesn't because one is an AssignPre and the other an AssignDly. Perhaps we should convert them all to Assign at some point.  (Proof of your point about simplifying node types in use a week ago.)

wsnyder, 2022-01-16:
Note also a similar issue could be constructed using any NodeAssign (AssignPre, AssignDly, Assign, AssignPost, etc?).