verilator/verilator #3281: Unable to run configure script in a single threaded environment

Choc13, 2022-01-18:
I'm working in an environment where `libc++` and `libc++abi` have been built without support for threads. 

When I try and checkout `v4.216` of Verilator and build this from source I hit a configure error which states that `"Verilator requires a C++11 or newer compiler.`". When I trace this back through the `configure.ac` script I see that on [line 485](https://github.com/verilator/verilator/blob/434c3c3ef3ae751b0a9f2d5934c2d2e986e65e81/configure.ac#L485) it is generating a test program that looks something like this:

```c
#include <thread>
#if (__cplusplus < 201103L)
# error "Too old"
#endif

int main()
{
    return 0;
}
```

The particular issue I am facing is that `<thread>` does not exist on my system because `libc++` was not compiled with threading support enabled. 

However, I have patched `configure.ac` locally to remove the `#include <thread>` and things seem to then work fine. After looking through the history of this part of `configure.ac` it seems that this [commit](https://github.com/verilator/verilator/commit/698e0fbbd1e866dc8abaa4900a1b56ca7316378b) led to this change. My understanding of this is that prior to that change it was only looking for C++11 support if threading was enabled, in which case it made sense to check if the `<thread>` header existed on the system. It now seems that C++11 support is required unconditionally and so I'm wondering if this check should now omit the call to `#include <thread>` and instead just check the `__cplusplus` macro value only?

I'm happy to submit my patch as a PR for this if it's a genuine issue. However, if I've misunderstood something then I'd be curious to know if building Verilator on a single threaded environment is something that is suppported.

What OS and distribution are you using?

Alpine 3.14.2 via a Docker image

wsnyder, 2022-01-18:
I believe thread is required in any compliant C++11 implementation, so yours is not compliant.

Verilator might work in a non-thread implementation if you don't use --threads, but there are no tests that check that, so it's not something that is likely to keep working.  Also "make test" is certain to fail.

So I think the default answer is the configure test is correct, this is failing intentionally, and we don't intend to support non-threaded C++11.  Or, not unless there's some extremely common use case and system that requires otherwise.  That is, what are you doing? ;)

Choc13, 2022-01-18:
Thanks for the response @wsnyder that's useful to know. 

We're building our own toolchain for a symbolic executor and right now it's based off a modified version of musl's libc in which we fake atomics. So we felt like it was safer when building libc++ to just disable threading altogether. So we're definitely not in a standard environment. It's actually possible that we can compile libc++ with threading enabled and then just disable it at a higher layer by not using `--threads` in Verilator like you say. 

Anyway, TLDR, I agree that your configure script is correct and we'll need to do something our side to work around it. Thanks again.

closed 2022-01-18 by Choc13