verilator/verilator #3283: Mac make test fail

quantrpeter, 2022-01-20:
```
~/workspace/verilator>gcc -v
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 13.0.0 (clang-1300.0.27.3)
Target: x86_64-apple-darwin21.2.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

```
~/workspace/verilator>g++ --version
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 13.0.0 (clang-1300.0.27.3)
Target: x86_64-apple-darwin21.2.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

```
Features (based on environment or compiled-in support):
    SystemC found      = 0
	perl /Users/chanshukping/workspace/verilator/test_regress/../bin/verilator --prefix Vt_a1_first_cc ../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_a1_first_cc -OD --debug-check --comp-limit-members 10 --debug --debugi 0 --gdbbt --no-dump-tree --trace --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_a1_first_cc t/t_a1_first_cc.v    > obj_vlt/t_a1_first_cc/vlt_compile.log
sh: gdb: command not found
-Info: --gdbbt ignored: gdb doesn't seem to be working
Starting Verilator 4.219 devel rev v4.216-95-g434c3c3e
Starting Verilator 4.219 devel rev v4.216-95-g434c3c3e
	make -C obj_vlt/t_a1_first_cc -f /Users/chanshukping/workspace/verilator/test_regress/Makefile_obj --no-print-directory VM_PREFIX=Vt_a1_first_cc TEST_OBJ_DIR=obj_vlt/t_a1_first_cc CPPFLAGS_DRIVER=-DT_A1_FIRST_CC  OPT_FAST=-O0 OPT_GLOBAL=-O0 Vt_a1_first_cc    > obj_vlt/t_a1_first_cc/vlt_gcc.log
driver: Entering directory '/Users/chanshukping/workspace/verilator/test_regress/obj_vlt/t_a1_first_cc'
g++  -I.  -MMD -I/Users/chanshukping/workspace/verilator/test_regress/../include -I/Users/chanshukping/workspace/verilator/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a1_first_cc___024root.h>" -DT_A1_FIRST_CC   -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp
g++    Vt_a1_first_cc__main.o verilated.o verilated_vcd_c.o Vt_a1_first_cc__ALL.a      -o Vt_a1_first_cc
Undefined symbols for architecture x86_64:
  "sc_time_stamp()", referenced from:
      vl_time_stamp64() in Vt_a1_first_cc__main.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Vt_a1_first_cc] Error 1
driver: Leaving directory '/Users/chanshukping/workspace/verilator/test_regress/obj_vlt/t_a1_first_cc'
%Warning: vlt/t_a1_first_cc: Exec of make failed: g++  -I.  -MMD -I/Users/chanshukping/workspace/verilator/test_regress/../include -I/Users/chanshukping/workspace/verilator/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a1_first_cc___024root.h>" -DT_A1_FIRST_CC   -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp

vlt/t_a1_first_cc: %Error: Exec of make failed: g++  -I.  -MMD -I/Users/chanshukping/workspace/verilator/test_regress/../include -I/Users/chanshukping/workspace/verilator/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a1_first_cc___024root.h>" -DT_A1_FIRST_CC   -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp
vlt/t_a1_first_cc: FAILED: Exec of make failed: g++  -I.  -MMD -I/Users/chanshukping/workspace/verilator/test_regress/../include -I/Users/chanshukping/workspace/verilator/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a1_first_cc___024root.h>" -DT_A1_FIRST_CC   -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:01
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:01

======================================================================
	#vlt/t_a1_first_cc: %Error: Exec of make failed: g++  -I.  -MMD -I/Users/chanshukping/workspace/verilator/test_regress/../include -I/Users/chanshukping/workspace/verilator/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a1_first_cc___024root.h>" -DT_A1_FIRST_CC   -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp
		make && test_regress/t/t_a1_first_cc.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Unsup 0  Time 0:01
make: *** [smoke-test] Error 10
```

thanks

jotego, 2022-01-20:
I'm also trying to get Verilator to run on a MacOS (v 12 Monterey version, ARM64). I didn't try the test build but I went directly for a simulation I had already running on linux. I still haven't got it working but got further in the compilation process with two changes:

I had to add `#include <array>` at the top of verilated_heavy.h
I had to change the PERL definition in verilated.mk:

```
#PERL = /opt/local/bin/perl5.28
PERL = /usr/bin/perl
```

Monterey ships with perl 5.30.

I still have a problem with the compile process as it tries to archive a .tmp file that doesn't exist:

```
ar -rcs Vjtopl__ALL.a @Vjtopl__ALL.a.tmp
```

which originates in verilated.mk too

wsnyder, 2022-01-20:
You must be using an old version, as verilated_heavy is not relevant. Please try with the latest release or github master.

jotego, 2022-01-20:
Thank you, @wsnyder. I tried the latest git following the install instructions and it worked without modifying any file. 

I also tried the `make test` after compiling and, as reported in this issue, it failed with the same message.

quark17, 2022-01-20:
> ```
> Undefined symbols for architecture x86_64:
>   "sc_time_stamp()", referenced from:
>       vl_time_stamp64() in Vt_a1_first_cc__main.o
> ```

This error was encountered in #3031.  In one of the comments, @wsnyder suspected that the macos tools were not supporting weak symbols.  This can be fixed by giving the flag `-undefined dynamic_lookup`, which I was able to confirm by manually adding it to `LDFLAGS`.  In the last comment, @wsnyder suggested that the right fix is to test for this flag in `configure.ac` and have the script add it to LDFLAGS (if the linker accepts it).  I've been meaning to make that change, but haven't gotten around to it, but would be happy if someone else got to it first.

samuel-falvo-wdc, 2022-07-12:
I forgot to mention that I was attempting to build the v4.224 tag.

wsnyder, 2022-07-12:
Try Verilating with "--CFLAGS -DVL_TIME_STAMP64" which should remove the reference.  If you do that make sure your code is using the new contextp->time() methods - ideally try making your code clean with "--CFLAGS -DVL_NO_LEGACY".

samuel-falvo-wdc, 2022-07-14:
Sorry, I forgot to respond until now.  Before you responded with your suggestion, and after I had submitted my report, I discovered @timsnyder 's workaround, which is here: https://github.com/regro-cf-autotick-bot/verilator-feedstock/commit/8cfa497c4fb5d33932384853a95ba442cf2422a3 .  I set (and exported) the environment variable prior to `./configure && make` and it seemed to compile in a way which let me run the tests.

When you write, "Try Verilating with...", what does that mean?  Are these configurations passed to `./configure`, or does this go some place else?  Thanks for your help!

wsnyder, 2022-07-14:
I mean add that flag to when you run the "verilator" executable.  It will then put the given flag into the makefile.  There's many other alternatives to get it to the compiler somehow including env vars...

Alex-Mann, 2022-09-15:
I also experienced this issue building on an M1 Macbook Air, and doing `export LDFLAGS="$LDFLAGS -undefined dynamic_lookup"` fixed this for me. Where would the proper fix go for this?

wsnyder, 2022-12-23:
Fixed with #3823.

closed 2022-12-23 by wsnyder