verilator/verilator #3285: --threads support for M1 CPUs

jotego, 2022-01-20:
It looks like the current thread implementation is specific to Intel CPU's. I get the following message when compiling with --threads on an M1 CPU computer:

```
In file included from /usr/local/share/verilator/include/verilated_threads.h:47:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/13.0.0/include/cpuid.h:14:2: error: this header is for x86 only
#error this header is for x86 only
```

I am quite surprised at the speed of Verilator using a single thread on my M1 CPU, so it would be amazing to boost it with more threads. Please consider supporting ARM/M1 CPUs for threads too.

wsnyder, 2022-01-20:
Verilator works on ARM under Ubuntu. It looks like this might be yet another case where Apple broke stuff (I assume you are on MacOS?).  Please try the git master, and if still breaks perhaps you can suggest a pull request to fix it appropriately?

jotego, 2022-01-21:
I tried version 4.218, which I suppose isn't much behind the current master. I can try the master one too. I do not have the skills to help with the code changes. Yet, I can help with testing.

wsnyder, 2022-01-22:
Thanks for checking. You'll need to wait until someone with a Mac fixes this.  Note in general we would appreciate a volunteer to take care of the Mac.

chenguokai, 2022-01-22:
Encountered the same problem on M1 Pro with Xcode 13. I figured out that `include/verilated_threads.h` included `cpuid.h` only for profile:
```c
#elif defined(__APPLE__)
        vluint32_t info[4];
        __cpuid_count(1, 0, info[0], info[1], info[2], info[3]);
        // info[1] is EBX, bits 24-31 are APIC ID
        if ((info[3] & (1 << 9)) == 0) {
            return -1;  // no APIC on chip
        } else {
            return (unsigned)info[1] >> 24;
        }
```

I can bypass this error by checking `__arm64__` macro:
```c
#if (defined(__APPLE__)) && (!defined(__arm64__))
# include <cpuid.h>  // For __cpuid_count()
#endif

// some code that we does not care
// in class VlProfileRec
#elif defined(__APPLE__) && !defined(__arm64__)
        vluint32_t info[4];
        __cpuid_count(1, 0, info[0], info[1], info[2], info[3]);
        // info[1] is EBX, bits 24-31 are APIC ID
        if ((info[3] & (1 << 9)) == 0) {
            return -1;  // no APIC on chip
        } else {
            return (unsigned)info[1] >> 24;
        }

```

wsnyder, 2022-01-22:
Great, can you wrap this into a pull request please?

Note there's some extra parens in your #if defined's that can get removed.

chenguokai, 2022-01-22:
> can you wrap this into a pull request please?
I will try
> Note there's some extra parens in your #if defined's that can get removed.
Extra parents are borrowed from some headers written by Apple, will remove if you like :)

wsnyder, 2022-01-23:
Appreciate your work to fix this.

closed 2022-01-23 by wsnyder