verilator/verilator #3287: AstAssignPre is optimizable into initialization

gezalore, 2022-01-21:
Currently, Verilator translates non blocking assignments into the folowing general sequence:

```
void _eval() {
   vlSyms->__Vdly__x = vlSyms->x;
    <possible, but not necesasry assignments to vlSyms->__Vdly__x>;
   vlSyms->x = vlSyms->__Vdly_x;
}
````

There are no assignments to the `__Vdly__*` signals anywhere outside of `_eval`, so the initial `__Vdly__x = x;` (the `AstAssignPre` created in V3Delayed) is redundant except on the very first invocation. We can instead execute the `__Vdly__x = x;` at the end of the initial settle phase, and subsequently remove it from `_eval`, saving one variable copy.

Note: Currently `__Vdly__x` might have been converted into a function local, which is no longer possible with this optimization as we would then rely on the `__Vdly__*` variables being persistent across invocations of `_eval`.

Note: Right now, non blocking assignments to arrays are implemented differently (as if they were memories) and this does not apply.

Note: If the variable `x` is also assigned via blocking assignments, then we need to keep the current behaviour.

wsnyder, 2022-01-22:
In most cases the ordering and later optimizations will both make this a local (as you note), but also remove one or both of the assignments using __Vdly_.  I suspect for best performance the optimization you suggest should maybe be done only if the assign still persists after both those opportunities for removal.  We'll need some experiments to prove which of the various options there is best.

gezalore, 2022-01-22:
FWIW (note to self) the localisation and removal would stil be possible _after_ this optimization, by simply assuming that the `_d` equals the `_q` at the beginning of the cycle (which was so far explicit due to the in loop AstAssignPre), which does hold unless there is a blocking assignment to the `_q` somewhere.

gezalore, 2026-09-03:
Still relevant and would also eliminate the need for using the "Unique flag" scheme in suspendables