verilator/verilator #3289: Compile error when trying to print an integer

oharboe, 2022-01-21:
@nekronos

Verilog:

```
module FIFO(
  input          clock,
  input          reset,
  output [511:0] io_master_writeData,
  output [63:0]  io_master_writeMask,
);
  always @(posedge clock) begin
     $fwrite(32'h80000002, "writeData=%x, writeMask=%x\n", io_master_writeData, io_master_writeMask);
  end
endmodule
```

To reproduce error:

```
verilator --version
Verilator 4.210 2021-07-07 rev v4.204-73-g8e2ba6a00
verilator --cc -Wno-fatal --top FIFO FIFO.v
cd obj_dir
make  -f VFIFO.mk VFIFO__ALL.a verilated.o verilated_fst_c.o verilated_threads.o
/nix/store/ag052gyghwrip8zlldk5d0qmhx182zq2-perl-5.34.0/bin/perl /nix/store/h5k0qd053wy9w2kba0l5ngw2bvbhsv0y-verilator-4.210/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VFIFO.cpp VFIFO___024root.cpp VFIFO___024root__Slow.cpp VFIFO__Syms.cpp > VFIFO__ALL.cpp
[deleted]
g++  -I.  -MMD -I/nix/store/h5k0qd053wy9w2kba0l5ngw2bvbhsv0y-verilator-4.210/share/verilator/include -I/nix/store/h5k0qd053wy9w2kba0l5ngw2bvbhsv0y-verilator-4.210/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o VFIFO__ALL.o VFIFO__ALL.cpp
In file included from VFIFO__ALL.cpp:4:
VFIFO___024root.cpp: In function 'void VFIFO___024root___sequent__TOP__1(VFIFO___024root*)':
VFIFO___024root.cpp:16:48: error: request for member 'data' in 'vlSelf->VFIFO___024root::io_master_writeData', which is of non-class type 'WData [16]' {aka 'unsigned int [16]'}
   16 |                512,vlSelf->io_master_writeData.data(),
      |                                                ^~~~
make: *** [/nix/store/h5k0qd053wy9w2kba0l5ngw2bvbhsv0y-verilator-4.210/share/verilator/include/verilated.mk:255: VFIFO__ALL.o] Error 1
```

Did you try it with the git master version?

What OS and distribution are you using?

NixOS

May we assist you in trying to fix this in Verilator yourself?

I think I'm out of my depth...

oharboe, 2022-01-21:
I can reproduce the problem on Ubuntu with Verilator 4.210.

So... problem solved. Use a newer version of Verilator than comes with NixOS out of the box.

oharboe, 2022-01-21:
Reopening in case the test-case is useful.

wsnyder, 2022-01-22:
Thanks for checking the newest version.  I think this test is fairly similar to an existing one, so closing.

closed 2022-01-22 by wsnyder