verilator/verilator #3290: Missing warning for different bit lengths

jotego, 2022-01-21:
I just found a bug in my RTL because this line had a 12 instead of a 13:

`
freq       = { 7'd0, fnum } + { {12{pm_offset[3]}}, pm_offset };
`

Surprisingly, Verilator linter is fine with a 12 and a 13. It will only issue warnings for numbers below 12 and above 13. The correct bit count occurs for 13. Is this behavior correct?

Here is the full, if short, module:

```
module jtopl_pg_inc (
    input        [ 2:0] block,
    input        [ 9:0] fnum,
    input signed [ 3:0] pm_offset,
    output reg   [16:0] phinc_pure
);

reg [16:0] freq;
/* verilator lint_on WIDTH */

always @(*) begin 
    freq       = { 7'd0, fnum } + { {12{pm_offset[3]}}, pm_offset };
    // Add PM here
    freq       = freq << block;
    phinc_pure = freq >> 1;
end

endmodule // jtopl_pg_inc
```

wsnyder, 2022-01-21:
This is correct, because it is common behavior for code to either want the carry bit, or not want the carry bit, and potentially make different assumptions on each side of the addition. Hence to avoid a lot of false warnings it accepts either.

jotego, 2022-01-21:
I understand the logic behind now, thank you.