verilator/verilator #3292: Python wrapper for DPI library

mmatzev, 2022-01-23:
It would by nice to be able to use the generated libraries also in a python environment.

Therefore, I wrote a simple small wrapper around the DPI library. The handwritten version works nice but needs updates if I modify the model input or output parameter. An automatic generation would be nice.

I see there different options:
a) verilator generates a systemVerilog file for the DPI lib. A small python script could convert this file in a python package as an add-on if the user requests it. Might be a little bit trick as it needs to parse the sv file. Small modifications in verilator might break the script. 
b) verilator generates a python wrapper like the systemVerilog wrapper for the DPI lib. This would need an additional function in V3ProtectLib.cpp to do the generation but could use the same infrastructure as the SV wrapper.
Draw back: There come next week someone who what's a next language. This makes verilator more complex and needs also maintenance.

If this feature is accepted I can do the work. 
Do you what this feature?
How (option a, b or another way) should it be implemented?

wsnyder, 2022-01-23:
Sounds interesting! I think I understand but can you please provide a pointer to the script or example output?

Is there a reason why this script might be only related to verilator? I would think in theory it could be made to support ant DPI simulator? If so, that suggests it perhaps it might be better left independent?

mmatzev, 2022-01-23:
As I said, currently it is only handwritten but I would like to start to write a generator. Hand written first try is attached.

Proposal would be: 
- Python class with properties for all inputs and outputs 
- Method: __init__, del and eval like in the C interface
- Automatic type recognition for basic types like logic, int, logic[], real, ...

For me it is unclear how the time information for the $time() function is handled in this case!?
For me it is unclear how or if it is possible to add tracing?

I think it is hard to make it universal. 

I had also a look to cocotb: Might be okay, but does not support real values and is more a framework than just a simple integration.
I had also a look to PyVerilator: Seems to be quite old and not (active) maintained, does not support real values, does not use the new feature of verilator to generate a library.
My idea would be to keep the maintance low but using the existing --lib-create function and a lightweight ctypes wrapper.

[python sv-rnm.zip](https://github.com/verilator/verilator/files/7920478/python.sv-rnm.zip)

Background of this idea:
We need to build models of the analog circuits as SV-RNM for the digital verification. We are going to check these models against our analog circuits but we would like to check these models also against the system level model in python of the datapath.

mmatzev, 2022-01-23:
Small note, as I tried to solve the task with cocotb before: cocotb supports real values but not in combination with verilator. Background: https://github.com/cocotb/cocotb/issues/2280

wsnyder, 2022-01-23:
My suggestion would be we add information to the XML output, or add a new XML output to have the information you need, then your script take it from there.  That will make it more universal if other people have similar requirements.  If long term this proves popular we could consider integrating the more pure python.

As to cocotb/cocotb#2280 no issue was filed here. Fixing the vpiString and vpiReal should be straight forward, perhaps you'd like to file a bug, and ideally later a pull request to do that?  It's a small fraction of the work (hours?) of what you're suggesting here.

mmatzev, 2022-01-23:
Hi, I have no go good overview - I'm more a user - , but I will try to file a related bug and to fix it. https://github.com/cocotb/cocotb/issues/2280 

About the XML output, just to understand you right. The approach would be to generate the library and the xml output (currently running verilator two times, if I see it right) and to use the xml output to generate the required wrapper files.
Is there a matching order of the function arguments from the xml and the lib file, which can be used? 
I would give it a try.

wsnyder, 2022-01-23:
You might need two runs currently, but we could probably dump and Verilate in one  assuming you only need certain inormation, with some changes.

wsnyder, 2022-02-10:
Closing as suspect you got your answers, feel free to follow up.

closed 2022-02-10 by wsnyder