verilator/verilator #3293: Error while building Verilator on Cygwin on x86_64 Windows 7

meetm81193, 2022-01-23:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
I am getting below error when trying to run below command. Also I tried running only make command but is giving same error.
I checked prerequisite like bison, flex, g++, python, perl, numactl.

make -j `nproc`
------------------------------------------------------------
making verilator in src
make -C src
make[1]: Entering directory '/home/Meet/verilator/src'
make -C obj_dbg -j 1  TGT=../../bin/verilator_bin_dbg.exe VL_DEBUG=1 -f ../Makefile_obj serial
make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg.exe VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj serial_vlcov
make -C obj_opt -j 1  TGT=../../bin/verilator_bin.exe -f ../Makefile_obj serial
make[2]: Entering directory '/home/Meet/verilator/src'
make[2]: warning: -j1 forced in submake: resetting jobserver mode.
make[2]: Entering directory '/home/Meet/verilator/src'
make[2]: warning: -j1 forced in submake: resetting jobserver mode.
make[2]: Entering directory '/home/Meet/verilator/src/obj_dbg'
make[2]: Nothing to be done for 'serial_vlcov'.
make[2]: Leaving directory '/home/Meet/verilator/src/obj_dbg'
make[2]: Nothing to be done for 'serial'.
make[2]: Leaving directory '/home/Meet/verilator/src/obj_opt'
make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg.exe VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj
make[2]: Nothing to be done for 'serial'.
make[2]: Leaving directory '/home/Meet/verilator/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_bin_dbg.exe VL_DEBUG=1 -f ../Makefile_obj
make -C obj_opt       TGT=../../bin/verilator_bin.exe -f ../Makefile_obj
make[2]: Entering directory '/home/Meet/verilator/src/obj_dbg'
make[2]: Entering directory '/home/Meet/verilator/src/obj_dbg'
make[2]: Entering directory '/home/Meet/verilator/src/obj_opt'
ccache g++  -Og -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../VlcMain.cpp -o VlcMain.o
      Compile flags:  g++ -Og -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/usr/local/share/verilator"
ccache g++  -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../Verilator.cpp -o Verilator.o
      Compile flags:  g++ -Og -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/usr/local/share/verilator"
ccache g++  -Og -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../Verilator.cpp -o Verilator.o
      Compile flags:  g++ -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/usr/local/share/verilator"
ccache g++  -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Active.cpp -o V3Active.o
g++.exe: error: /home/Meet/.ccache/tmp/V3Active.tmp.Meet_Mehta.1442.ii: No such file or directory
g++.exe: fatal error: no input files
compilation terminated.
make[2]: *** [../Makefile_obj:308: V3Active.o] Error 1
make[2]: *** Waiting for unfinished jobs....
ccache g++  -Og -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Active.cpp -o V3Active.o
g++.exe: error: /home/Meet/.ccache/tmp/V3Active.tmp.Meet_Mehta.1448.ii: No such file or directory
g++.exe: fatal error: no input files
compilation terminated.
make[2]: *** [../Makefile_obj:308: V3Active.o] Error 1
make[2]: *** Waiting for unfinished jobs....
g++.exe: error: /home/Meet/.ccache/tmp/Verilator.tmp.Meet_Mehta.1440.ii: No such file or directory
g++.exe: fatal error: no input files
compilation terminated.
make[2]: *** [../Makefile_obj:308: Verilator.o] Error 1
make[2]: Leaving directory '/home/Meet/verilator/src/obj_opt'
make[1]: *** [Makefile:60: ../bin/verilator_bin.exe] Error 2
make[1]: *** Waiting for unfinished jobs....
g++.exe: error: /home/Meet/.ccache/tmp/Verilator.tmp.Meet_Mehta.1441.ii: No such file or directory
g++.exe: fatal error: no input files
compilation terminated.
make[2]: *** [../Makefile_obj:308: Verilator.o] Error 1
make[2]: Leaving directory '/home/Meet/verilator/src/obj_dbg'
make[1]: *** [Makefile:66: ../bin/verilator_bin_dbg.exe] Error 2
g++.exe: error: /home/Meet/.ccache/tmp/VlcMain.tmp.Meet_Mehta.1439.ii: No such file or directory
g++.exe: fatal error: no input files
compilation terminated.
make[2]: *** [../Makefile_obj:308: VlcMain.o] Error 1
make[2]: Leaving directory '/home/Meet/verilator/src/obj_dbg'
make[1]: *** [Makefile:70: ../bin/verilator_coverage_bin_dbg.exe] Error 2
make[1]: Leaving directory '/home/Meet/verilator/src'
make: *** [Makefile:141: verilator_exe] Error 2

What 'verilator --version' are you using?  Did you try it with the git master version?
Unable to build verilator

What OS and distribution are you using?
Cygwin 64
Windows 7

May we assist you in trying to fix this in Verilator yourself?
Yes please i need solution for this.

wsnyder, 2022-01-23:
>++.exe: error: /home/Meet/.ccache/tmp/V3Active.tmp.Meet_Mehta.1442.ii: No such file or directory

Something is broken with your ccache setup, or more likely you are out of disk space.  Either way, this isn't anything Verilator itself needs to fix.   Please report back what it was to help others.

meetm81193, 2022-01-23:
So the Drive C:\ in which cygwin64 is installed does have currenlty few gb of empty space.

I tried to update ccache using below command.
apt-cyg update ccache

After updating, I am still having that error. Would you tell me how much Disk Space I will be required for verilator currently i am having 6gb free space.

Also what do you mean by ccache system and what could be the possible solution for it.

wsnyder, 2022-01-23:
CCache is a completely separate program that speeds up the compiler.  

If you "touch /home/Meet/.ccache/tmp/V3Active.tmp" does it work or not?

You can disable it by uninstalling it, or try "export CCACHE_DISABLE=true"

meetm81193, 2022-01-24:
No file named V3Active.tmp is available.

So i tried other option of disabling it with the export command and it is working perfectly.

Now the new problem which is arrising because of missing FlexLexer.h file as see below
making verilator in src
make -C src
make[1]: Entering directory '/home/Meet/verilator/src'
make -C obj_dbg -j 1  TGT=../../bin/verilator_bin_dbg.exe VL_DEBUG=1 -f ../Makefile_obj serial
make[2]: Entering directory '/home/Meet/verilator/src/obj_dbg'
make[2]: Nothing to be done for 'serial'.
make[2]: Leaving directory '/home/Meet/verilator/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_bin_dbg.exe VL_DEBUG=1 -f ../Makefile_obj
make[2]: Entering directory '/home/Meet/verilator/src/obj_dbg'
      Compile flags:  g++ -Og -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/usr/local/share/verilator"
ccache g++  -Og -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -Wno-char-subscripts -Wno-unused -c ../V3ParseLex.cpp -o V3ParseLex.o
In file included from ../V3ParseLex.cpp:28:0:
V3Lexer_pregen.yy.cpp:368:10: fatal error: FlexLexer.h: No such file or directory

          ^
compilation terminated.
make[2]: *** [../Makefile_obj:313: V3ParseLex.o] Error 1
make[2]: Leaving directory '/home/Meet/verilator/src/obj_dbg'
make[1]: *** [Makefile:66: ../bin/verilator_bin_dbg.exe] Error 2
make[1]: Leaving directory '/home/Meet/verilator/src'
make: *** [Makefile:141: verilator_exe] Error 2

I am having version flex 2.6.4 and i also have FlexLexer.h file in /usr/include/ folder.

Would you please help me whats the issue here?

wsnyder, 2022-01-24:
Google for the error message and "cygwin".  First hit suggests try installing libfl-dev.

wsnyder, 2022-02-13:
Closing as assuming you got this cleared up.

closed 2022-02-13 by wsnyder