verilator/verilator #3294: signed number operation

RaynardQiao, 2022-01-25:
version：  master version

code:
```
module twocounter(in0, in1, out);
  input [7:0] in0;
  input [7:0] in1;
  output  [7:0] out;
  wire signed [15:0] mult_tmp;
  wire signed [7:0] in1;
  wire signed [7:0] in0;
  assign mult_tmp = $signed({1'b0, in0}) * in1;      // Error code，  in1  is  signed variable， so  this operator should be  signed multiplication

  assign out= mult_tmp[7:0];
endmodule

```

obj_dir:
```
VL_INLINE_OPT void Vtwocounter___024root___combo__TOP__1(Vtwocounter___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtwocounter__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtwocounter___024root___combo__TOP__1\n"); );
    // Body
    vlSelf->out = (0xffU & ((IData)(vlSelf->in0) * (IData)(vlSelf->in1)));      // This result is not Signed operation
}

```

Signed number multiplication becomes unsigned number multiplication。

wsnyder, 2022-02-10:
I suspect as $signed gets eliminated in some cases it probably looses the conversion.  Needs more research as to fix.

RaynardQiao, 2022-02-14:
I was trying to fix the problem and found that in the prelinkdot-find.tree phase, the $signed information was lost.
The V3LinkDot.cpp file is modified as follows, in function void visit(AstVar* nodep)  line1094; The signed information is checked and saved.
```
                    bool ansiBad = findvarp->isAnsi() || nodep->isAnsi();  // dup illegal with ANSI
                    if (ansiBad || nansiBad) {
                        bool ansiWarn = ansiBad && !nansiBad;
                        if (ansiWarn) {
                            static int didAnsiWarn = false;
                            if (didAnsiWarn++) ansiWarn = false;
                        }
                        nodep->v3error("Duplicate declaration of signal: "
                                       << nodep->prettyNameQ() << '\n'
                                       << (ansiWarn ? nodep->warnMore()
                                                          + "... note: ANSI ports must have"
                                                            " type declared with the I/O (IEEE "
                                                            "1800-2017 23.2.2.2)\n"
                                                    : "")
                                       << nodep->warnContextPrimary() << '\n'
                                       << findvarp->warnOther()
                                       << "... Location of original declaration\n"
                                       << findvarp->warnContextSecondary());
                        // Combining most likely reduce other errors
                        findvarp->combineType(nodep);
                        findvarp->fileline()->modifyStateInherit(nodep->fileline());
                    } else {
                        findvarp->combineType(nodep);

                        // For issue3294 
                        if (nodep->getChildDTypep()->numeric().isSigned() && 
                        !findvarp->getChildDTypep()->numeric().isSigned())
                        {
                            findvarp->getChildDTypep()->numeric(VSigning(true));
                        }

                        findvarp->fileline()->modifyStateInherit(nodep->fileline());
```

The modified code passes the test_regress test.

wsnyder, 2022-02-14:
Great, thanks for looking into fixing this. Can you please post a pull request for this?  I think the same code should be in both if() branches, so I would suggest make a new function that is called in both places with this code:

```
                        findvarp->combineType(nodep);
                        findvarp->fileline()->modifyStateInherit(nodep->fileline());
                        if (nodep->getChildDTypep()->numeric().isSigned() && 
                        !findvarp->getChildDTypep()->numeric().isSigned()) {
                            findvarp->getChildDTypep()->numeric(VSigning(true));
                        }
```

wsnyder, 2022-02-16:
Much appreciate your pull request to fix this.

closed 2022-02-16 by wsnyder