verilator/verilator #3298: Where are *___sequent__TOP__x defined?

Choc13, 2022-02-01:
I'm able to successfully compile and run a Verilog project called [iob-cache](https://github.com/IObundle/iob-cache) using Verilator. I'm not trying to debug it, but I'm struggling to find the definitions of some of the C++ code that has been generated.

In a file called Viob_cache___024root__DepSet_h62dfe991__0.cpp I see several declarations like the following:

```cpp
void Viob_cache___024root___sequent__TOP__7(Viob_cache___024root* vlSelf);
void Viob_cache___024root___combo__TOP__8(Viob_cache___024root* vlSelf);
```

The stacktrace in my debug session shows that it's called `Viob_cache___024root___sequent__TOP__7` and so I was trying to locate the source, but I cannot find its definition anywhere. Is this function being code generated somehow? 

I was the under the impression that the code gen part only happened when calling the `verilator` executable and that all of the generated source was dumped into the `obj_dir` folder before being built and linked with the C++ compiler. However, I only see the declaration of this function in there and not its definition after the call to `verilator` has finished. So I'm confused as to where the definition for this function is located / when it is generated.

Thanks.

gezalore, 2022-02-01:
They are in the similarly named C++ files. They are forward declared where used, and defined in some unspecified order in the cpp files (this is to avoid having header dependencies which sometimes speeds up recompilation via ccache with small code changed). Grep the generated output for the signature without the trailing semicolon and you should find the definition.

Choc13, 2022-02-01:
Thanks for the fast response @gezalore. I had been trying this expecting to find them defined elsewhere. I've just checked again and I had annoyingly had an option set in VSCode that was only grepping through open files. Sorry about that. Found it now.

closed 2022-02-01 by Choc13