verilator/verilator #3299: Internal error when assigning integer array parameter

michael-platzer, 2022-02-04:
Hi,

when attempting to assign the values of an unpacked integer array parameter directly I get an internal error "InitArray type not assigned by AstPattern/Var visitor".

Code example:
```
module mymod
    #(
        parameter int unsigned VAL[2] = '{1, 2}
    )
    ();
endmodule

module top
    ();
    mymod #(
        .VAL ( '{1, 2} )
    ) myinst ();
endmodule
```

Verilator version and error message:
```
$ verilator --version
Verilator 4.218 2022-01-17 rev UNKNOWN.REV
$ verilator -Wall --cc array_params.sv --top-module top --lint-only
%Error: Internal Error: array_params.sv:11:16: ../V3Width.cpp:2239: InitArray type not assigned by AstPattern/Var visitor
                                                             : ... In instance top
   11 |         .VAL ( '{1, 2} )
      |                ^~
```

Declaring the array as a localparam first and passing that as the parameter works fine though.

wsnyder, 2022-02-10:
Confirmed.

yodalee, 2022-04-28:
@michael-platzer Can you provide the example with localparam that can pass compilation?

michael-platzer, 2022-04-29:
@yodalee Sure, below is a modified version that uses a localparam instead and passes compilation:

```
module mymod
    #(
        parameter int unsigned VAL[2] = '{1, 2}
    )
    ();
endmodule

module top
    ();

    localparam int unsigned LOCAL[2] = '{1, 2};
    mymod #(
        .VAL ( LOCAL )
    ) myinst ();
endmodule
```

closed 2022-10-21 by wsnyder