verilator/verilator #3300: Internal error in V3Param.cpp: Could not find array index in unlinked text

daviddavo, 2022-02-06:
I'm trying to compile a file that has a generated array of interfaces, but I get the following error with Verilator 4.216

```terminal
$ verilator -V
Verilator 4.216 2021-12-05 rev UNKNOWN.REV

Copyright 2003-2021 by Wilson Snyder.  Verilator is free software; you can
redistribute it and/or modify the Verilator internals under the terms of
either the GNU Lesser General Public License Version 3 or the Perl Artistic
License Version 2.0.

See https://verilator.org for documentation

Summary of configuration:
  Compiled in defaults if not in environment:
    SYSTEMC            =
    SYSTEMC_ARCH       =
    SYSTEMC_INCLUDE    =
    SYSTEMC_LIBDIR     =
    VERILATOR_ROOT     = /usr/share/verilator
    SystemC system-wide = 1

Environment:
    MAKE               =
    PERL               =
    SYSTEMC            =
    SYSTEMC_ARCH       =
    SYSTEMC_INCLUDE    =
    SYSTEMC_LIBDIR     =
    VERILATOR_ROOT     =
    VERILATOR_BIN      =

Features (based on environment or compiled-in support):
    SystemC found      = 1
$ make
rm -rf *.log *.s *.hex *.dis *.tbl irun* vcs* simv* *.map \
       verilator* *.exe obj* *.o ucli.key vc_hdrs.h csrc \
	   *.csv work dataset.asdb library.cfg
verilator --cc -Wall --assert --build -j -CFLAGS  common_defines.vh include/noc_types.sv \
	-Isrc \
	-f tools/flist sim/tb_noc.sv \
    --top tb_noc --autoflush
%Error: Internal Error: src/mesh.sv:77:70: ../V3Param.cpp:1060: Could not find array index in unlinked text: 'west_down__BRA__0__KET__' for node: CELLARRAYREF 0x55b626649330 <e5907#> {i77cs}  ports_up__Viftop
                                         : ... In instance tb_noc.mesh
   77 |                                  .down(nodes_h[i].nodes_w[j].ports_up[EAST]),
      |                                                                      ^
make: *** [Makefile:18: verilator-build] Error 1
```

You can see the file that's causing the error here:
https://github.com/daviddavo/tfg_poc_noc/blob/07caa05b6ea06983e252be1e97a397127cb7348f/src/mesh.sv#L47-L86

```systemverilog
   generate
      genvar          i, j;
      
      // Generating nodes
      // TODO: Flatten into one array
      for ( i = 0; i < MESH_HEIGHT; i++ ) begin: nodes_h
         for ( j = 0; j < MESH_WIDTH; j++ ) begin: nodes_w
            // TODO: Use `NODE_PORTS
            node_port ports_up[4]();
            node_port ports_down[4]();
            
            node #(
                   .X(i+1),
                   .Y(j+1),
                   .X_EDGE(MESH_HEIGHT),
                   .Y_EDGE(MESH_WIDTH)
                   ) node (
                           .clk(clk),
                           .rst(rst),
                           .ports_up(ports_up),
                           .ports_down(ports_down)
                           );
         end
      end
      
      // Connects node to next node horizontally
      for ( i = 0; i < MESH_HEIGHT; i++) begin
         for ( j = 0; j < MESH_WIDTH-1; j++) begin
            // East to West
            node_link east2west (
                                 .down(nodes_h[i].nodes_w[j].ports_up[EAST]),
                                 .up(nodes_h[i].nodes_w[j+1].ports_down[WEST])
                                 );
            // West to East
            node_link west2east (
                                .down(nodes_h[i].nodes_w[j+1].ports_up[WEST]),
                                .up(nodes_h[i].nodes_w[j].ports_down[EAST])
                                );
         end
      end
```

wsnyder, 2022-02-06:
Perhaps it's confused by the multiple indicies.  Can you make this into a minimized example which fails please?

wsnyder, 2022-05-12:
We can reopen if get a standlone test case.

saicarpio, 2024-07-23:
Hello. I'm working on a similar proyect and got the same error. I've tried to write a simplified example that fails with the same error msg.

```systemverilog

`define MAX 4

interface node_port;
    logic data;
    modport down (input data);
    modport up (output data);
endinterface

module link (
    node_port.down down,
    node_port.up up
);
    assign up.data = down.data;
endmodule

module dut (
    node_port.down port_down[`MAX],
    node_port.up port_up[`MAX]
);
    genvar i;
    generate
        for (i = 0; i < `MAX; i++)
            assign port_up[i].data = port_down[(i+2)%`MAX].data;
    endgenerate

endmodule

module dut_array(
    node_port.down port_down [`MAX],
    node_port.up port_up     [`MAX]
);
    
    genvar i, j;
    generate
        for (i = 0; i < `MAX ; i++) begin: nodes
            node_port port_up[`MAX]();
            node_port port_down[`MAX]();

            dut dut(.port_up(port_up), .port_down(port_down));
        end

        for (i = 0; i < `MAX - 1; i++ ) begin
            link n2s (.down(nodes[i+1].port_up[0]),
                      .up(nodes[i].port_down[2]));

            link s2n (.down(nodes[i].port_up[2]),
                      .up(nodes[i+1].port_down[0]));
            
            link e2w (.down(nodes[i+1].port_up[1]),
                      .up(nodes[i].port_down[3]));

            link w2e (.down(nodes[i].port_up[3]),
                      .up(nodes[i+1].port_down[1]));
        end

        link nI (.down(port_down[0]),
                 .up(nodes[0].port_down[0]));

        link eI (.down(port_down[1]),
                 .up(nodes[3].port_down[1]));
        
        link sI (.down(port_down[2]),
                 .up(nodes[3].port_down[2]));

        link wI (.down(port_down[3]),
                 .up(nodes[0].port_down[3]));

        link nO (.up(port_up[0]),
                 .down(nodes[0].port_up[0]));

        link eO (.up(port_up[1]),
                 .down(nodes[3].port_up[1]));
        
        link sO (.up(port_up[2]),
                 .down(nodes[3].port_up[2]));

        link wO (.up(port_up[3]),
                 .down(nodes[0].port_up[3]));

    endgenerate

endmodule

module tb_top;
    bit clk;

    always #10 clk = ~clk;

    node_port port_down[`MAX]();
    node_port port_up[`MAX]();

    dut_array duta(
        .port_down(port_down), 
        .port_up(port_up)
    );

  initial begin
    $finish;
  end
endmodule

```
It's a single file called **fail.sv**  which fails with the following command

```
$ verilator --binary fail.sv --top tb_top
%Error: Internal Error: fail.sv:43:47: ../V3Param.cpp:1287: Could not find array index in unlinked text: 'port_down__BRA__3__KET__' for node: CELLARRAYREF 0x62b9f6f2a760 {d43bv} @dt=0@  port_up
                                     : ... note: In instance 'tb_top.duta'
   43 |             link n2s (.down(nodes[i+1].port_up[0]),
      |                 
```

In my case, I'm working with Verilator 5.020
 ```
$ verilator -V
Verilator 5.020 2024-01-01 rev (Debian 5.020-1)

Copyright 2003-2024 by Wilson Snyder.  Verilator is free software; you can
redistribute it and/or modify the Verilator internals under the terms of
either the GNU Lesser General Public License Version 3 or the Perl Artistic
License Version 2.0.

See https://verilator.org for documentation

Summary of configuration:
  Compiled in defaults if not in environment:
    SYSTEMC            = 
    SYSTEMC_ARCH       = 
    SYSTEMC_INCLUDE    = 
    SYSTEMC_LIBDIR     = 
    VERILATOR_ROOT     = /usr/share/verilator
    SystemC system-wide = 1

Environment:
    MAKE               = 
    PERL               = 
    SYSTEMC            = 
    SYSTEMC_ARCH       = 
    SYSTEMC_INCLUDE    = 
    SYSTEMC_LIBDIR     = 
    VERILATOR_BIN      = 
    VERILATOR_ROOT     = 

Supported features (compiled-in or forced by environment):
    COROUTINES         = 1
    SYSTEMC            = 1
```
I'm no expert in neither SystemVerilog, nor Verilator.