verilator/verilator #3301: Missing LATCH from always_comb and case

obfuscated, 2022-02-09:
It seems the LATCH warning is not emitted for case statements. Only for ifs.
This is an example file:
```
module aw(input logic a, output logic b, output logic c );
   always_comb begin
       case (a)
           1'b0: b=1;
       endcase
   end
   always_comb begin
       if (a==1'b0)
           c=1;
   end
endmodule
```
And this is the result from `Verilator 4.218 2022-01-17 rev UNKNOWN.REV`

```
> verilator test_always_comb.sv --lint-only
%Warning-CASEINCOMPLETE: test_always_comb.sv:4:8: Case values incompletely covered (example pattern 0x1)
    4 |        case (a)
      |        ^~~~
                         ... For warning description see https://verilator.org/warn/CASEINCOMPLETE?v=4.218
                         ... Use "/* verilator lint_off CASEINCOMPLETE */" and lint_on around source to disable this message.
%Warning-LATCH: test_always_comb.sv:10:4: Latch inferred for signal 'c' (not all control paths of combinational always assign a value)
                                        : ... Suggest use of always_latch for intentional latches
   10 |    always_comb begin
      |    ^~~~~~~~~~~
%Error: Exiting due to 2 warning(s)
```

I'm new to verilog and hardware design, but I've discovered this problem after Vivado's synthesis tools started to complain on similar code and verilator didn't.
BTW, why is this even allowed to compile without an error? Isn't the idea/goal of always_comb to force combinational circuits and thus no latches.

OS: Gentoo Linux

> May we assist you in trying to fix this in Verilator yourself?

No, time at the moment sorry.

wsnyder, 2022-02-10:
The LATCH warning was specifically disabled on case statements due to certain restrictions in the analysis of the case statement.  I suspect when there's a CASEINCOMPLETE however, we know that there must be a LATCH so should not suppress this.

agrobman, 2022-02-12:
BTW, this code can create different results with different simulators and if used in real design have unpredictable behavior.
( the output of this module can be 'X' on standard simulators and 0 with verilator if _a_ is X or 1)

obfuscated, 2022-02-12:
@agrobman What do you mean? This is something I don't want to write. It is something I consider an error and I hope that verilator would detect it and help me fix it.

closed 2025-09-29 by wsnyder