verilator/verilator #3302: How to use V{prefix}__ALL.a library

dbear496, 2022-02-10:
I have in the past used verilator with the --exe option to directly create an executable from a top-level verilog module and a C++ file with a main function. However, now I want my C++ program to connect to two separate top-level modules at the same time. I quickly found that this cannot be done with the --exe option since verilator handles only one top-level module at a time. So I ran verilator twice -- once for each top-level module -- to create two V{prefix}__ALL.a libraries that my C++ program can link to. Unfortunately, I cannot figure out how to properly compile and link my C++ manually, and there appears to be no documentation on this. When I naively compile with `g++ main.cpp VModA__ALL.a VModB__ALL.a`, I get an error that 'verilated.h' is not found, and if I include `$VERILATOR_ROOT/include`, then I get a long string of linker errors. When I used the --exe option in the past, I noticed my C++ program was compiled with quite a lot of g++ options that I don't fully understand, so I am unsure how to compile/link my C++ program manually.

Is there a recommended way to manually compile/link C++ programs for verilator? Is there a recommended format for a custom makefile that will call the verilator makefiles and compile/link the C++ program? Am I even approaching this properly? Perhaps some more documentation is in order, since all the documentation/examples/tutorials that I could find use the --exe option.

wsnyder, 2022-02-10:
There's two options.  Roughly:

A: Pass the first module's archive to the second Verilator run.
  1. verilator VmodA.
  2. make -C obj_dir VModA.mk VModA__ALL.a
  3. verilator VmodA. --exe VModA__ALL.a main.cpp
  4. make -C obj_dir VModB.mk

B. g++ requires you to compile and link separately (AFAIK) when using libraries.  So first compile your .cpp file with "-c" to make an object. Then make the verilator library objects (there is no pretty make-an-archive rule for this currently), then link that .o along with the .a's in another g++ run.
  1. verilator VmodA
  2. verilator VmodB
  3. make -C obj_dir VModA.mk VModA__ALL.a
  4. make -C obj_dir VModB.mk VModB__ALL.a
  5. my.mk needs roughly:
       main.cpp: main.o
            # This will run  g++ -c -o main.o main.cpp
       main: main.o $(VK_GLOBAL_OBJS)
             # The dependencies above will build main.o and verilated*.o
             g++ -o main main.o $(VK_GLOBAL_OBJS) VModA__ALL.a VModB__ALL.a 
  6. make -C obj_dir ../my.mk

There's no example for this as generally if you're doing something that complicated you'll need to understand makefiles anyways. (Sorry, we welcome pull requests to document things like this better.)

Note also there's a test that does this which is test_regress/t/t_trace_two_dump_cc.pl, but the test makefiles are slightly different and aren't very easy to read.

closed 2022-02-10 by wsnyder

dbear496, 2022-02-10:
Thanks for the prompt reply. I tried to get option B to work, but I always seemed to end up with linking errors. Option A seems to have worked.