verilator/verilator #3303: Portable way to mix enums visible in C++ and DontCare values?

obfuscated, 2022-02-10:
I want to have an enum visible in C++ and I also want to initialize variables of this type with dont-cares. What is the portable way to do it?
I've tried 2 things and they both fail in one of the tools I want to use.

Here is an example:
```
package pkg;
    typedef enum logic [1:0] {
        E0=2'b00,
        E1=2'b01,
        E2=2'b10,
        E_dont_care=2'b??
    } MyE /*verilator public*/;
endpackage

module m(input logic[1:0] a, output logic b, output logic c );
    pkg::MyE e;
    always_comb begin
        // e=2'b??; // <- works in verilator, fails in vivado's synthesis tool (it doesn't allow non-enum values to be used)
        e=pkg::E_dont_care; // <- works in vivado, fails in verilator
        case (a)
            2'b00: e=pkg::E0;
            2'b01: e=pkg::E1;
        endcase
    end

    logic [1:0] test;
    assign test=e;
    assign b=test[0];
    assign c=test[1];
endmodule
```

The C++ main.cpp file:
```
#include "verilated.h"
#include "Vtest_always_comb.h"

int main() {
	return 0;
}
```

The verilator command and the error it produces is below:
```
> verilator -Wall -Wpedantic -Wno-fatal test_always_comb.sv  --exe --build --cc main.cpp
%Warning-DECLFILENAME: test_always_comb.sv:1:9: Filename 'test_always_comb' does not match PACKAGE name: 'pkg'
    1 | package pkg;
      |         ^~~
                       ... For warning description see https://verilator.org/warn/DECLFILENAME?v=4.218
                       ... Use "/* verilator lint_off DECLFILENAME */" and lint_on around source to disable this message.
%Warning-CASEINCOMPLETE: test_always_comb.sv:15:9: Case values incompletely covered (example pattern 0x2)
   15 |         case (a)
      |         ^~~~
%Error: Internal Error: test_always_comb.sv:6:21: ../V3EmitCConstInit.h:103: 4-state value in constant pool
    6 |         E_dont_care=2'b??
      |                     ^~~~~
```
I suppose this error is expected, because enum in C doesn't support 4-state values, but error message is pretty obscure, I've discovered that the problem is caused the presence of the public comment only after trying to make a simpler example to post a question.

wsnyder, 2022-02-10:
Verilator is completely 2-state during runtime.  So this won't be supported.

You are right the error could be better, but it would be similar, "4-state value isn't allowed in public enum".  That's a very low priority to improve, so instead closing.

BTW I suspect what would satisfy synthesis is this, which is the more common way to write this:

```
 always_comb begin
        unique case (a)
            2'b00: e=pkg::E0;
            2'b01: e=pkg::E1;
            default: e='x;
        endcase
    end
```

Note also the common methodology is ? is used as a test, e.g. is bit anything, while X is used on the right value side meaning don't care.

closed 2022-02-10 by wsnyder

obfuscated, 2022-02-10:
The problem is that the sample I posted is minimized for demonstration purpose. The actual code is comprised of multiple levels of case statements (it is a cpu control module, I have an outer case for the opcode and inner cases for functXX fields) and actually there are multiple enumerated signals to set, so if I do it like this it will become too verbose and error prone.

I know about the difference between x and ?, this was a copy-paste typo in the example, sorry about that.

Would it be possible to implement a hint to hide certain enumerations  from the C++ side (/*verilator hidden*/ or /*verilator private*/)? I suppose this would allow the example to be compiled without any modifications.

I've tried to use a cast assignment statement, but it also failed. Is it possible to use something like this?

wsnyder, 2022-02-11:
I think what you're asking for is to pass all the non-4-state values into the public.  This is committed.