verilator/verilator #3309: VCD corruption

drom, 2022-02-16:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

I am running complex simulation and VCD dump is broken in multiple places. Very similar scenario. Here are 3 examples:

## example 1

```
b00 'o"
b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b0000000000000000000000000000000000000000000000000000000000000000b00000000000000000000000000000000000000000000000000000000000000000ho"
0io"
```

Note that long line 2 contains multiple `b00...` substrings concatenated.

`ho"` is defined as: `$var wire  1 ho" auto_tile_reset_domain_tile_bpwatch_out_0_valid_0 $end`

***

## example 2

```
b000000000000000 '[;
b00000000000000000000b00000/[;
00[;
```
Similar issue in line 2

`/[;` is defined as: `$var wire  1 /[; out_f_woready_104 $end`

***

## example 3

```
0@zD
001CzD
1DzD
```
`CzD` is a scalar defined as `$var wire  1 CzD stateUpd $end`

***

```
> verilator --version
Verilator 4.210 2021-07-07 rev v4.210
```

I tried `git master` version. Cant compile the design at all. Will file another issue.

Tried on OSs:

```
Linux zzz 5.16.5-1-default #1 SMP PREEMPT Thu Feb 3 05:26:48 UTC 2022 (1af4009) x86_64 x86_64 x86_64 GNU/Linux
```

```
Linux zzz 4.18.0-305.19.1.el8_4.x86_64 #1 SMP Tue Sep 7 07:07:31 EDT 2021 x86_64 x86_64 x86_64 GNU/Linux
```

wsnyder, 2022-02-17:
I don't think there's been a recent fix for this sort of issue.  It looks like the buffer is corrupting somehow, but we'd need an example that reproduces it, or you'd need to debug it with assistance.

drom, 2022-02-17:
@wsnyder thank you for the reply.
Yes, looks like this issue exists on range of version at least from v4.100
I am trying to create smaller example that would trigger the issue.

drom, 2022-02-17:
One interesting observation:

It looks like signals in VCD dumped in sorted order:

```
...
b00000000000000000000000000000000 H,
b00000000000000000000000000000000 I,
b00000000000000000000000000000000 J,
b00000000000000000000b00000R,
0S,
0T,
...
```

Some IDs are missing. For example:

`H,`  `I,`  `J,` (missing items `K,` `L,` `M,` `N,` `O,` `P,` `Q,`)  `R,` `S,` `T,`

Corruption happens in place of missing items.

The VCD header defines all correct items but none of the potentially missing:

```
$var wire 32 H, progbuf0 [31:0] $end
$var wire 32 I, progbuf1 [31:0] $end
$var wire 32 J, data0 [31:0] $end
...
$var wire  1 R, ABSTRACTCSWrEn $end
```

drom, 2022-02-26:
Somehow the amount of corrupted places changes when removing `_` from temporary wires.

wsnyder, 2022-02-26:
Assuming it is some pointer corruption or similar, any small change in what is traced or in memory layout is likely to move the problem.

drom, 2022-02-26:
> Assuming it is some pointer corruption or similar, any small change in what is traced or in memory layout is likely to move the problem.

Yep, that is what happening.

SiFiveGregS, 2022-02-26:
As far as trying to replicate the issue on a simple toy input design, I have a question.  What Verilog design characteristics are needed in order to stimulate the generation of multiple traceInitSub<index>(userp, tracep) calls in traceInitTop?  (such as in the snippet below).  I'd like to join the effort to try to replicate the issue.  

What I'm seeing in the Verilated code I'm looking at is there are a lot more additional traceInitSubN methods present in the generated code (up to N=41), but only N=0..6 seem to be actually getting reached in the call chain from traceInitTop (however some of the signals in the non-reached init functions are later trying to be emitted).  So I'm hoping to replicate that sort of pattern on a simple design (however contrived or trivial the design might be).  Just not sure how to structure a toy design to get the multiple traceInitSubN functions to get generated.

void VTestDriver::traceInitTop(void* userp, VerilatedVcd* tracep) {
    VTestDriver__Syms* __restrict vlSymsp = static_cast<VTestDriver__Syms*>(userp);
    VTestDriver* const __restrict vlTOPp VL_ATTR_UNUSED = vlSymsp->TOPp;
    // Body
    {
        vlTOPp->traceInitSub0(userp, tracep);
        vlTOPp->traceInitSub1(userp, tracep);
        vlTOPp->traceInitSub2(userp, tracep);
        vlTOPp->traceInitSub3(userp, tracep);
        vlTOPp->traceInitSub4(userp, tracep);
        vlTOPp->traceInitSub5(userp, tracep);
        vlTOPp->traceInitSub6(userp, tracep);
    }
}

wsnyder, 2022-02-28:
I don't think Verilator makes "traceInitSub" any longer, have you tried using the git master version?

SiFiveGregS, 2022-02-28:
Thanks, I had been trying 4.204, because the Verilated output I was looking at was generated from that version.  But if I can get my hands on the original input design that corresponds to the output I'm looking at, I'll try running the same design through git master version and proceed from there.

wsnyder, 2022-03-07:
Note in latest git master, if you compile the verilated code with "-DVL_DEBUG" it will assert() on the case that I think was causing your assertion.  Of course this isn't a real fix but at least it isn't silent.

drom, 2022-03-08:
> Note in latest git master, if you compile the verilated code with "-DVL_DEBUG" it will assert() on the case that I think was causing your assertion. Of course this isn't a real fix but at least it isn't silent.

Trying right away.

drom, 2022-03-09:
When compiled with `-DVL_DEBUG` it core dumps like this:

```
INFO(../src/verilator-main.cpp): Random seed is 1
INFO(../src/verilator-main.cpp): dump VCD to dhrystone.verilator.vcd
TestFinisher instance path is top.TestDriver.testHarness.system.testIndicator.i_TF.f80

VTestDriver: /sifive/verilator/v4.216-134-gae005434/share/verilator/include/verilated_vcd_c.cpp:611: void VerilatedVcd::finishLine(vluint32_t, char *): Assertion `suffixp[0]' failed.
Aborted (core dumped)
```

wsnyder, 2022-03-10:
That's consistent with the issue you described, something not getting declared properly.

wsnyder, 2022-04-12:
At least one case that has been identified to cause this is when an interface has another interface inside it.  I'm working on a verilation-time assertion (vs runtime) that was causing this issue in general, and a fix for the interface-in-interface issue.

wsnyder, 2022-04-14:
Believe this is now fixed, please try it out, thanks.

closed 2022-04-14 by wsnyder

drom, 2022-04-19:
Tested new verilator. VCD corruption is gone