verilator/verilator #3313: fscanf return 0 on EOF (instead of -1)

jotego, 2022-02-18:
If $fscanf is repeatedly called it will eventually return 0 when the EOF is reached, but according to Verilog 2001 documentation, it should return -1 upon EOF.

[image]

The image is from the book _Verilog-2001: A Guide to the New Features of the Verilog HDL_

Tested on Verilator 4.199 devel rev v4.110-32-g9483ebefa

wsnyder, 2022-02-18:
Makes sense.  Would you be willing to make a pull request to test this in t_sys_file_basic.v, and fix the source in include/verilated.h?

Please also test and fix if needed $fgetc, $ftell, $frewind, $fseek.

jotego, 2022-02-18:
What you want from me is to fix the t_sys_file_basic.v file, which is a test bench, so that it will detect this problem, right?
I don't know what yo do with include/verilated.h. I don't see the reference to fscanf there.

wsnyder, 2022-02-18:
Right, add a test for those transactions against a closed handle to t_sys_file_basic.  The other file is verilated.cpp ans search for e.g. FSCANF.

wsnyder, 2022-03-05:
Any luck working on this pull request?

jotego, 2022-03-07:
Sorry, I think this goes beyond my capabilities.

closed 2022-03-07 by wsnyder

jotego, 2022-03-08:
I saw that you closed it. Is it because it is fixed now?

wsnyder, 2022-03-08:
This is fixed.