verilator/verilator #3315: Open to PR to fail execution if an `x` value arrives at an `if` statement?

hughperkins, 2022-02-20:
How may we help - what is your question?

The verilog standard states that an `if` statement which receives a condition of `x` should evaluate the `else` branch, as though it had received a `0` condition. This means that the following code, when simulated, will convert an `x` value into a `0` value:

```
if (foo) assign out = 0;
else assign out = 1;
```
However, when synthesized, an effective value of `foo` of `x` will actually give an output value of `x`. Therefore the simulation will lead to different behavior from synthesis.

What are standard solutions to this problem? I'm thinking of creating a PR that would add an option to verilator such that at runtime, if an `if` statement received an `x` value, this would cause a runtime error.

Thoughts?

wsnyder, 2022-02-20:
First, Verilator is a 2 state simulator, so no X can be used at runtime.

Some other simulators have X pessimism, whereby with an X input both sides ate evaluated and conflicts become x's.

There are also formal tools (none open source AFAIK) that can look for  missing resets etc, this is really the best solution.

instead Verilator supports randomizing x's, see the docs. This should find all reset bugs like you asked about.

closed 2022-02-20 by wsnyder

hughperkins, 2022-02-20:
> Verilator is a 2 state simulator, so no X can be used at runtime.

Ok, interesting. Thanks!

> There are also formal tools (none open source AFAIK) that can look for missing resets etc, this is really the best solution.

Ok.

> instead Verilator supports randomizing x's, see the docs. This should find all reset bugs like you asked about.

Ok. Thanks!

hughperkins, 2022-04-03:
> instead Verilator supports randomizing x's, see the docs. This should find all reset bugs like you asked about.

This works very nicely! Thanks! Found a bug that iverilog failed to find :) Nice :)