verilator/verilator #3320: Improve performance of non-blocking assignments to associative arrays

gezalore, 2022-02-24:
Associative arrays are currently implemented as a `std::map`, and non-blocking assignments use a complete `__Vdly_` copy, this results in two complete copies of the `std::map` per `_eval`, which can be very slow for large arrays and dominate the simulation time. The ideas in #3288 would be applicable here and improve performance significantly.

udif, 2022-02-24:
Is there a reason not to use std::unordered_map instead?
Are we talking about Verilator code, or actual Verilator-generated runtime code?

wsnyder, 2022-02-24:
Note BTW that unordered_map is much slower to init than map for small maps due to making the hash table.  Unfortunately.

closed 2024-04-13 by wsnyder