verilator/verilator #3321: Duplicate declaration of block: 'unnamedblk1'

drom, 2022-02-26:
While compiling the following file:

```verilog
module top #(
  parameter type STATUS_REG_t = bit [31:0]
)(
  input STATUS_REG_t status_regs [1],
  input bit clock,
  input bit reset
);
TestFinisher t1(
  .status_regs(status_regs),
  .clock(clock),
  .reset(reset)
);
endmodule

interface TestFinisher #(
  parameter type STATUS_REG_t = bit [31:0]
) (
  input STATUS_REG_t status_regs [1],
  input bit clock,
  input bit reset
);

  initial begin
    string instance_path = $sformatf("%m");
    $display("TestFinisher instance path is %s\n", instance_path);
  end

  bit [0:0] pending;
  always_comb begin
    foreach (status_regs [ index ] ) begin
      pending[index] = 1;
    end
  end

endinterface : TestFinisher
```

By running `verilator --lint-only test.sv`

`Verilator 4.219 devel rev v4.216-111-ge52a4ac7` from the master.

I am getting this error:

```
%Error: test.sv:31:5: Duplicate declaration of block: 'unnamedblk1'
                    : ... In instance top
   31 |     foreach (status_regs [ index ] ) begin
      |     ^~~~~~~
        test.sv:24:11: ... Location of original declaration
   24 |   initial begin
      |           ^~~~~
%Error: test.sv:31:5: Duplicate declaration of CELLINLINE 'unnamedblk1': 'unnamedblk1'
   31 |     foreach (status_regs [ index ] ) begin
      |     ^~~~~~~
        test.sv:24:11: ... Location of original declaration
   24 |   initial begin
      |           ^~~~~
%Error: Exiting due to 2 error(s)
```

I am running:

`Linux linux-sydi 5.16.8-1-default #1 SMP PREEMPT Thu Feb 10 11:31:59 UTC 2022 (5d1f5d2) x86_64 x86_64 x86_64 GNU/Linux`

Willing to assist debugging this issue.

drom, 2022-03-04:
Can someone look at this issue?

closed 2022-03-05 by wsnyder

Zvord, 2023-01-18:
The same error still happens in another case:
```systemverilog
module Foo;
   function void func();
      int a[2];

      begin
         int t;
      end

      foreach (a[i]) begin
      end
   endfunction
endmodule
```
Shall this issue be reopened, or it's better to create a new one?

Here's output from `verilator --lint-only ../bad.v --debug --debugi 5`. I've marked with **!!!!** the place where "unnamedblk1" name assignment happens. 
```
- V3LinkDot.cpp:207:  LinkDotState: 
- V3LinkDot.cpp:1509: LinkDotFindVisitor: 
- V3LinkDot.cpp:911:  Module not under any CELL or top - dead module: MODULE 0x154004f20 {c1ai}  Foo  L0 [1ps]
- V3LinkDot.cpp:870:       Link Module: MODULE 0x154004f20 {c1ai} u1=0x154004d80  Foo  L0 [1ps]
- V3LinkDot.cpp:1057:    TASK 0x154005100 {c2as}  func
- V3LinkDot.cpp:1023:    BEGIN 0x154005a80 {c5ah} [UNNAMED]
!!!!
- V3LinkDot.cpp:1023:    BEGIN 0x154006280 {c9aw} [UNNAMED]
- V3Ast.cpp:1122:     Dumping obj_dir/Vbad_004_prelinkdot-find.tree
- V3LinkDot.cpp:1704: LinkDotParamVisitor: 
- V3LinkDot.cpp:1556:    MODULE 0x154004f20 {c1ai} u1=0x154004d80 u4=0x1  Foo  L0 [1ps]
- V3Ast.cpp:1122:     Dumping obj_dir/Vbad_005_prelinkdot-param.tree
- V3LinkDot.cpp:3346: LinkDotResolveVisitor: 
- V3LinkDot.cpp:3098:    TASK 0x154005100 {c2as} u1=0x154004e20  func
- V3LinkDot.cpp:3084:    BEGIN 0x154005a80 {c5ah} u1=0x1540065a0  unnamedblk1 [UNNAMED]
- V3LinkDot.cpp:3090:    cur=se0x1540065a0
- V3LinkDot.cpp:3095:    cur=se0x154004e20
- V3LinkDot.cpp:3120:    FOREACH 0x1540063e0 {c9ah} u1=0x1540066e0
- V3LinkDot.cpp:3084:    BEGIN 0x154006280 {c9aw} [UNNAMED]
- V3LinkDot.cpp:3095:    cur=se0x1540066e0
- V3LinkResolve.cpp:513:linkResolve: 
- V3LinkLValue.cpp:360:linkLValue: 
- V3LinkJump.cpp:322: linkJump: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vbad_009_linkjump.tree
- V3LinkInc.cpp:292:  linkIncrements: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vbad_010_linkinc.tree
- V3Param.cpp:1311:   param: 
- V3LinkDot.cpp:3397: linkDotParamed: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vbad_012_prelinkdot.tree
- V3LinkDot.cpp:207:  LinkDotState: 
- V3LinkDot.cpp:1509: LinkDotFindVisitor: 
- V3LinkDot.cpp:911:  Module not under any CELL or top - dead module: MODULE 0x154004f20 {c1ai}  Foo  L0 [1ps]
- V3LinkDot.cpp:870:       Link Module: MODULE 0x154004f20 {c1ai} u1=0x1540072e0  Foo  L0 [1ps]
- V3LinkDot.cpp:1057:    TASK 0x154005100 {c2as}  func
- V3LinkDot.cpp:1023:    BEGIN 0x154005a80 {c5ah}  unnamedblk1 [UNNAMED]
- V3LinkDot.cpp:1023:    BEGIN 0x154006280 {c9aw} [UNNAMED]
- V3Ast.cpp:1122:     Dumping obj_dir/Vbad_013_prelinkdot-find.tree
- V3LinkDot.cpp:1704: LinkDotParamVisitor: 
- V3LinkDot.cpp:1556:    MODULE 0x154004f20 {c1ai} u1=0x1540072e0 u4=0x1  Foo  L0 [1ps]
- V3Ast.cpp:1122:     Dumping obj_dir/Vbad_014_prelinkdot-param.tree
- V3LinkDot.cpp:3346: LinkDotResolveVisitor: 
- V3LinkDot.cpp:3098:    TASK 0x154005100 {c2as} u1=0x154007380  func
- V3LinkDot.cpp:3084:    BEGIN 0x154005a80 {c5ah} u1=0x154007500  unnamedblk1 [UNNAMED]
- V3LinkDot.cpp:3090:    cur=se0x154007500
- V3LinkDot.cpp:3095:    cur=se0x154007380
- V3LinkDot.cpp:3120:    FOREACH 0x1540063e0 {c9ah} u1=0x154007650
- V3LinkDot.cpp:3084:    BEGIN 0x154006280 {c9aw} [UNNAMED]
- V3LinkDot.cpp:3095:    cur=se0x154007650
- V3Ast.cpp:1122:     Dumping obj_dir/Vbad_015_linkdotparam.tree
- V3Dead.cpp:538:     deadifyModules: 
- V3Width.cpp:7179:   width: 
[... 51 lines cut ...]
-node: BEGIN 0x154008a50 {c9ah} u1=0x154009c40  unnamedblk1 [UNNAMED] [IMPLIED]
- V3Ast.cpp:1122:     Dumping obj_dir/Vbad_029_linkdot-preerr.tree
- V3LinkDot.cpp:1023:    BEGIN 0x154006280 {c9aw} [UNNAMED]
- V3LinkDot.cpp:870:       Link Module: MODULE 0x154004f20 {c1ai} u1=0x15400a030 u4=0x1  Foo  L0 [1ps]
- V3LinkDot.cpp:1057:    TASK 0x154005100 {c2as} u1=0x154009960  func
- V3LinkDot.cpp:1023:    BEGIN 0x154005a80 {c5ah} u1=0x154009b00  unnamedblk1 [UNNAMED]
- V3LinkDot.cpp:1023:    BEGIN 0x154008a50 {c9ah} u1=0x154009c40  unnamedblk1 [UNNAMED] [IMPLIED]
- V3LinkDot.cpp:271:  name unnamedblk1
- V3LinkDot.cpp:272:  Var1 BEGIN 0x154008a50 {c9ah} u1=0x15400a520  unnamedblk1 [UNNAMED] [IMPLIED]
- V3LinkDot.cpp:273:  Var2 BEGIN 0x154005a80 {c5ah} u1=0x15400a310  unnamedblk1 [UNNAMED]
%Error: ../bad.v:9:7: Duplicate declaration of block: 'unnamedblk1'
                    : ... In instance Foo
    9 |       foreach (a[i]) begin
      |       ^~~~~~~
        ../bad.v:5:7: ... Location of original declaration
    5 |       begin
      |       ^~~~~
```

Do we have here two instances of a Visitor, and each instance starts to numerate blocks from 1?

```
./bin/verilator_bin --version                              
Verilator 5.005 devel rev v5.004-65-g3fe81a383 (mod)
```

wsnyder, 2023-01-18:
Please create a new one, thanks.