verilator/verilator #3324: Assert: iterateAndNext node has no back

ldoolitt, 2022-02-26:
Regression introduced (thank you, git bisect!) on commit 8c58612,
Dec 22, 2021: "Improve V3Inline speed and memory consumption".

On some production code, I see:
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
So of course I add those flags.  The output is pretty chatty (and I have a full log saved), but I think the important part is

```
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Starting Verilator 4.217 devel rev v4.216-78-g8c58612a3
[...]
- V3Ast.cpp:1130:     Dumping obj_dir/Vinjector_025_unknown.tree
- V3Inline.cpp:740:   inlineAll:
Program received signal SIGSEGV, Segmentation fault.
AstNode::iterateAndNext (this=0x1, v=...) at /redacted/verilator/src/V3Ast.cpp:832
832         UASSERT_OBJ(!(nodep && !nodep->m_backp), nodep, "iterateAndNext node has no back");
#0  AstNode::iterateAndNext (this=0x1, v=...) at /redacted/verilator/src/V3Ast.cpp:832
#1  0x0000555555791167 in AstNode::iterateChildren (this=this@entry=0x555559ecc8b0, v=...) at /redacted/verilator/src/V3Ast.cpp:806
#2  0x00005555559efa99 in VNVisitor::iterateChildren (nodep=0x555559ecc8b0, this=0x7fffffffd5b0) at /redacted/verilator/src/V3Ast.h:3009
#3  InlineIntfRefVisitor::visit (this=0x7fffffffd5b0, nodep=<optimized out>) at /redacted/verilator/src/V3Inline.cpp:695
#4  0x00005555557a6be4 in AstCell::accept (this=<optimized out>, v=...) at /redacted/verilator/src/V3AstNodes.h:2894
#5  0x00005555557910c1 in AstNode::iterateAndNext (this=<optimized out>, v=...) at /redacted/verilator/src/V3Ast.cpp:843
#6  0x0000555555791178 in AstNode::iterateChildren (this=this@entry=0x555558464710, v=...) at /redacted/verilator/src/V3Ast.cpp:807
#7  0x00005555559ed1ec in VNVisitor::iterateChildren (nodep=0x555558464710, this=<optimized out>) at /redacted/verilator/src/V3Ast.h:3009
#8  InlineIntfRefVisitor::visit (this=<optimized out>, nodep=<optimized out>) at /redacted/verilator/src/V3Inline.cpp:668
#9  0x00005555557a755c in AstNetlist::accept (this=<optimized out>, v=...) at /redacted/verilator/src/V3AstNodes.h:9327
#10 0x00005555559ec17b in VNVisitor::iterate (nodep=0x555556032360, this=0x7fffffffd5b0) at /redacted/verilator/src/V3Ast.h:3005
#11 InlineIntfRefVisitor::InlineIntfRefVisitor (nodep=0x555556032360, this=0x7fffffffd5b0) at /redacted/verilator/src/V3Inline.cpp:732
#12 V3Inline::inlineAll (nodep=0x555556032360) at /redacted/verilator/src/V3Inline.cpp:761
#13 0x0000555555763ba5 in process () at /redacted/verilator/src/V3Global.h:129
#14 0x0000555555764959 in verilate (argString=...) at /redacted/verilator/src/Verilator.cpp:609
#15 0x0000555555766bca in main (argc=126, argv=<optimized out>, env=<optimized out>) at /redacted/verilator/src/Verilator.cpp:735
```
I took the production code that triggered this bug, and boxed it in to a simple 600 kByte mash of 12 .sv files, 95 .v files,and 7 .vh files.  This has eliminated all the original project complexity of machine-generated code, Makefiles, and multiple git repos.
I posted a tarball of the "test case" to http://recycle.lbl.gov/~ldoolitt/fault_20220225.tar.bz2 . it's not exactly secret, just messy.
@wsnyder, you mentioned on Mar 19, 2021 https://github.com/verilator/verilator/issues/2842#issuecomment-802482797 you have a "testcase reducer"; maybe that can help me.
I could also mess around by hand and see if I can get any hints that way.

ldoolitt, 2022-02-26:
I removed 21 files and reduced the size of the compressed tarball by 23.6%.  I slip-streamed the new tarball to that same URL given above, now 79472 bytes.

More importantly, I may have isolated the issue. The error disappears when I disable a hardware feature that depends on a cute (and effective!) **recursive** Verilog construction.  I'll see if this insight lets me build up a much smaller test case.

ldoolitt, 2022-02-27:
Sho' nuff.  I again slip-streamed in a new tarball. This time it contains four .v files, 223 non-blank non-comment LOC, and the compressed tarball size is 5580 bytes. License on the Verilog files is good (BSD). Same Verilator fault signature. This could turn into a real test case.

Commentary about the upstream code is at https://github.com/BerkeleyLab/Bedrock/blob/master/dsp/banyan_readme.txt

ldoolitt, 2022-02-27:
Github doesn't accept .tar.bz2 files.  I attach a .zip version here for archival purposes.

[fault_20220225.zip](https://github.com/verilator/verilator/files/8147745/fault_20220225.zip)

gezalore, 2022-02-27:
Thanks for the example. I will try to take a look, but currently have very limited bandwidth.

ldoolitt, 2022-03-01:
It turns out that three of those four Verilog files are not essential to reproducing this issue.  Here's an even smaller zip file, containing one .v file and a README.  banyan.v is 100% unmodified production code, containing 70 non-blank non-comment lines-of-code.

Sounds simple, right?  The two modules defined therein recurse down from the top-level module, yielding seven instances of the first, and 41 (?) instances of the second.

The upstream repo includes a Verilog test bench suitable for running under Icarus Verilog.  In my Copious Free Time [TM] I'll try to port that to C++ in a way that could fit within the Verilator regression test framework.  That would be more convincing than a simple --lint-only exercise.

[issue_3324.zip](https://github.com/verilator/verilator/files/8158471/issue_3324.zip)

ldoolitt, 2022-03-03:
Here's an updated zip file with the functional test-bench described above.  Works with all the Verilator versions I've tried between v4.010 through 56f9d24.
[issue_3324b.zip](https://github.com/verilator/verilator/files/8174959/issue_3324b.zip)

gezalore, 2022-03-05:
Thanks for the neat test case, which passes with the fix just pushed. Please file a new issue if there are other ones.

closed 2022-03-05 by gezalore