verilator/verilator #3334: Installing Verilator configured with --enable-ccwarn breaks (some) production use cases

ldoolitt, 2022-03-03:
Steps to reproduce:

1. A gcc version subject to bug #69602
     https://gcc.gnu.org/bugzilla/show_bug.cgi?id=69602
This includes Debian Buster's gcc 8.3.0 and Debian Bullseye's gcc 10.2.1.

2. A Verilator test bench containing C code that triggers the above bug. Commonly seen compiling networking code (on Linux) that includes the construct
  `if (errno == EAGAIN || errno == EWOULDBLOCK)`
If you want a non-synthetic example, see https://github.com/BerkeleyLab/Bedrock in the badger/tests directory, where `make Vrtefi_blob` involves compiling ethernet_model.c in the Verilator context.

3. Build a modern Verilator from source, install, and use it for a build of the form described above.  No problem.

4. Rebuild and reinstall Verilator, this time with the --enable-ccwarn option provided to configure.  No warnings show up when compiling Verilator itself.  Good so far!  Try the above test case to exercise the resulting install.  Blam!
```
g++  -I. -Wall -Wextra -Wfloat-conversion -Wlogical-op -Werror -MMD -I/redacted/share/verilator/include -I/redacted/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o ethernet_model.o ../ethernet_model.c
../ethernet_model.c: In function ‘int ethernet_model(int, int, int*, int*, int)’:
../ethernet_model.c:99:26: error: logical ‘and’ of equal expressions [-Werror=logical-op]
    if ((errno != EAGAIN) && (errno != EWOULDBLOCK)) {
        ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[1]: *** [Vrtefi_blob.mk:66: ethernet_model.o] Error 1
make[1]: Leaving directory '/redacted/bedrock/badger/tests/obj_dir'
make: *** [Makefile:285: Vrtefi_blob] Error 2
```

Fix (IMHO):  move -Wlogical-op from CFG_CXXFLAGS_WEXTRA to CFG_CXXFLAGS_SRC in configure.ac.  This doesn't change the compile options used for building verilator_bin itself: -Wlogical-op is still there if --enable-ccwarn is specified at configure-time.  But that gcc warning flag is no longer added to application build steps.  The resulting Verilator installation becomes actually useful for production!

Otherwise, --enable-ccwarn should be marked as test-only, not for installation and general use.  Personally, I like the test-like-you-fly and fly-like-you-test approach to building code.  And the more checks the compiler can provide, the better. Hence the previous suggestion.

wsnyder, 2022-03-05:
But... this is pointing out warnings in *your* code, not anything Verilator is generating.  Why should we lower Verilator's warning standards for bad user code?  That would result in less good code from Verilator as once we disable the warning the code will rot and issues would creep in.

Or am I missing something?

ldoolitt, 2022-03-05:
I agree in theory.  But as the discussion in the gcc bug listed above concludes, all known rewrites of user code to avoid triggering this gcc complaint would actually make the user code worse.  That specifically includes the `(errno != EAGAIN) && (errno != EWOULDBLOCK)` case that I hit.  Ideally gcc would figure out that this is correct defensive programming.  Until [gcc bug #69602](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=69602) is fixed, the least-bad option seems to be turning off -Wlogical-op.

I took a quick stab at convincing obj_dir/foo.mk to add c++ flags of my own, but came up short.  That would let me  override -Wlogical-op with -Wno-logical-op when needed. If the lines in foo.mk that look like
```$(OBJCACHE) $(CXX) $(CXXFLAGS) $(CPPFLAGS) $(OPT_FAST) -c -o $@ $<```
had an additional entry like $(EXTRA_CXXFLAGS) that would give me a workaround. Is there an existing recipe for this?

wsnyder, 2022-03-05:
Does "-CFLAGS -Wno-logical-op" work?

ldoolitt, 2022-03-05:
Probably.  Give me ten minutes for another --enable-ccwarn build and I'll let you know for sure.

closed 2022-03-05 by ldoolitt