verilator/verilator #3341: How do we reference multiple verilog/RTL files for sub components to the top level rtl that need to be verilated?

amerhebi, 2022-03-08:
How may we help - what is your question?

In the hello world systemc example : examples/cmake_hello_sc/CMakeLists.txt
the make file references to 1 .v file as it is a simple example

# Add the Verilated circuit to the target
verilate(example SYSTEMC
  INCLUDE_DIRS "../make_hello_sc"
  SOURCES ../make_hello_sc/top.v)

The question is if let's say top.v (for large logic blocks) has multiple components that are instantiated IP1 , IP 2 and each IP is made up of multiple IP*_sub* .v files also in different directories from the one for top.v : 

for example 
directory : RTL_V
- RTL_V/top.v
directory IP1 : IP1_RTL_V
- IP1_RTL_V/IP1.v
- IP1_RTL_V/IP1_sub1.v
- IP1_RTL_V/IP1_sub2.v
- IP1_RTL_V/IP1_sub3.v
directory IP2 : IP2_RTL_V
- IP2_RTL_V/IP2.v
- IP2_RTL_V/IP2_sub1.v
- IP2_RTL_V/IP2_sub2.v
- IP2_RTL_V/IP2_sub3.v

Question: 
- Do we need to explicitly list all the .v files as below ?

# Add the Verilated circuit to the target
verilate(example SYSTEMC
  INCLUDE_DIRS "/RTL_DIR"
                          " /IP1_RTL_V"
                          " /IP2_RTL_V"
  SOURCES /RTL_DIR/top.v 
                  /IP1_RTL_V/IP1.v
                  /IP1_RTL_V/IP1_sub1.v
                  /IP1_RTL_V/IP1_sub2.v
                  /IP1_RTL_V/IP1_sub3.v
                  /IP2_RTL_V/IP2.v
                  /IP2_RTL_V/IP2_sub1.v
                  /IP2_RTL_V/IP2_sub2.v
                  /IP2_RTL_V/IP2_sub3.v)

or is it just enough to refer the RTL directories and top.v as below ?

verilate(example SYSTEMC
  INCLUDE_DIRS "/RTL_DIR"
                          " /IP1_RTL_V"
                          " /IP2_RTL_V"
  SOURCES /RTL_DIR/top.v)

wsnyder, 2022-03-11:
I'm not sure how cmake deals with these dependencies, @patstew ?

patstew, 2022-03-11:
You need to list the top module, and files containing interfaces as far as I remember. It should be equivalent to calling verilator from the command line.

closed 2022-03-12 by wsnyder