verilator/verilator #3344: Warn about undriven nets

jotego, 2022-03-08:
I have spent the last five days looking after a bug that basically consisted in a signal that was undriven. Intel Quartus does not warn about it neither does Verilator or NCVerilog. I wonder if it really makes sense not to warn when a signal is used in an expression but it is undriven. By _undriven_ I mean that it is declared as a `wire` but that's all, no assignments.

ldoolitt, 2022-03-08:
I see warnings like
```
Warning-UNDRIVEN: real_sim_mux.v:93: Signal is not driven: cavity_array_cav_elec_mode_2_out_couple_out_coupling_addr
```
even from Verilator 4.010. What version are you using? What happens if you add `-Wall` to your command line?

jotego, 2022-03-09:
I see... it actually exists but it is disabled by default. Using `-Wall` or `-Wwarn-UNDRIVEN` brings the warning up.

I will add that flag to the scripts. 
Thank you!

closed 2022-03-09 by jotego