verilator/verilator #3346: Compile issue in SV case-stmt with ++ or -- operators AND one of --trace or --coverage enabled

drewranck, 2022-03-09:
Version
Verilator 4.219 devel rev v4.218-40-g181b9a57

OS
Debian GNU/Linux 10  -- Chromebook linux 98.0.4758.107 (Official Build) (64-bit)

May we assist you in trying to fix this in Verilator yourself? Maybe :) I didn't see a similar issue, and I thought existing issues with various compilers had been addressed already, but perhaps this is some fringe issue with Debian g++.

Workarounds:
1. Refactor the SV code to avoid a case-stmt (if-else)
2. Refactor the SV code to avoid ++ or --, such as n = n + 1'b1;
3. Do not compile with --trace or --coverage

My local test_regress is:
t_case_default.v:
```
module t (/*AUTOARG*/
   // Inputs
   clk
   );

   input clk;

   integer cyc; initial cyc = 0;

   logic      push_valid, push_ready;
   logic      pop_valid, pop_ready;

   always @ (posedge clk) begin
      cyc <= cyc + 1;
      {push_valid, push_ready, pop_valid, pop_ready}
        <= 4'($urandom);

     if (cyc > 100) begin
	     $write("*-* All Finished *-*\n");
       $finish();
     end

   end

  // real RTL code snippet:

  logic [3:0] count_d;
  logic [3:0] count_q = '0;

  always_ff @(posedge clk) begin : flops
    count_q   <= count_d;
  end

  logic  push, pop;
  assign push = push_valid && push_ready;
  assign pop = pop_valid && pop_ready;

  // Uncomment one of these define lines to get Verilator to compile
//`define HACK_TO_PASS
//`define HACK_TO_PASS2

`ifdef HACK_TO_PASS

  always_comb begin : update_counts
    count_d = count_q;

    // refactored case stmt as if-else-if
    if (push && !pop) begin
      count_d++;
    end else if (!push && pop) begin
      count_d--;
    end
  end
`else

`ifdef HACK_TO_PASS2
  always_comb begin : update_counts
    count_d = count_q;

    case ({push, pop})
    2'b10: begin
      // refatored ++ as n + 1'b1.
      count_d = count_d + 1'b1;
    end
    2'b01: begin
      count_d = count_d - 1'b1;
    end
    default : ;
    endcase // case ({push, pop})
  end
`else

  // This is the one that fails:
  always_comb begin : update_counts
    count_d = count_q;

    case ({push, pop})
    2'b10: begin
      count_d++; // issue is here, case with ++, --
    end
    2'b01: begin
      count_d--;
    end
    default : ;
    endcase // case ({push, pop})
  end
`endif
`endif

endmodule

```

t_case_default.pl:
```
#!/usr/bin/env perl
if (!$::Driver) { use FindBin; exec("$FindBin::Bin/bootstrap.pl", @ARGV, $0); die; }
# DESCRIPTION: Verilator: Verilog Test driver/expect definition
#
# Copyright 2003 by Wilson Snyder. This program is free software; you
# can redistribute it and/or modify it under the terms of either the GNU
# Lesser General Public License Version 3 or the Perl Artistic License
# Version 2.0.
# SPDX-License-Identifier: LGPL-3.0-only OR Artistic-2.0

scenarios(simulator => 1);

compile(
    verilator_flags2 => ["--trace --Os -x-assign 0"],
    );

execute(
    check_finished => 1,
    );

ok(1);
1;

```

This is giving me some errors:
```
/usr/bin/perl /home/drewranck/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vt_case_default.cpp Vt_case_default___024root__DepSet_h0cedbb89__0.cpp Vt_case_default__Trace__0.cpp Vt_case_default___024root__Slow.cpp Vt_case_default___024root__DepSet_h0cedbb89__0__Slow.cpp Vt_case_default__Syms.cpp Vt_case_default__Trace__0__Slow.cpp > Vt_case_default__ALL.cpp
ccache g++  -I.  -MMD -I/home/drewranck/verilator/include -I/home/drewranck/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_case_default -DVM_PREFIX=Vt_case_default -DVM_PREFIX_INCLUDE="<Vt_case_default.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_case_default___024root.h>" -DT_CASE_DEFAULT   -O0 -c -o Vt_case_default__ALL.o Vt_case_default__ALL.cpp
In file included from Vt_case_default__ALL.cpp:4:
Vt_case_default___024root__DepSet_h0cedbb89__0.cpp: In function ‘void Vt_case_default___024root___sequent__TOP__0(Vt_case_default___024root*)’:
Vt_case_default___024root__DepSet_h0cedbb89__0.cpp:43:50: error: expected ‘;’ before ‘}’ token
         t__DOT__update_counts__DOT____Vincrement1} else if (
                                                  ^
                                                  ;
Vt_case_default___024root__DepSet_h0cedbb89__0.cpp:49:50: error: expected ‘;’ before ‘}’ token
         t__DOT__update_counts__DOT____Vincrement2}
                                                  ^
                                                  ;
In file included from Vt_case_default__ALL.cpp:7:
Vt_case_default___024root__DepSet_h0cedbb89__0__Slow.cpp: In function ‘void Vt_case_default___024root___settle__TOP__0(Vt_case_default___024root*)’:
Vt_case_default___024root__DepSet_h0cedbb89__0__Slow.cpp:39:50: error: expected ‘;’ before ‘}’ token
         t__DOT__update_counts__DOT____Vincrement1} else if (
                                                  ^
                                                  ;
Vt_case_default___024root__DepSet_h0cedbb89__0__Slow.cpp:45:50: error: expected ‘;’ before ‘}’ token
         t__DOT__update_counts__DOT____Vincrement2}
                                                  ^
                                                  ;
make: *** [/home/drewranck/verilator/include/verilated.mk:257: Vt_case_default__ALL.o] Error 1
```

wsnyder, 2022-03-10:
Sounds like you're willing to try a fix which would be appreciated.  First please make a test_regress test showing the issue from your test case.
Then to fix, in V3LinkInc Verilator converts ++/-- into simpler operations and part of that is determining where to insert.  I suspect this doesn't handle case items right.  If you run with "--debug --dump-treei 9" you should get a obj_dir/*linkinc*.tree file, showing how it transformed.  Fix the source code to properly transform it (you can see how it should look by e.g. using an if statement instead).

drewranck, 2022-03-10:
Thanks for the tip on where to look. I believe V3LinkInc.cpp has a missing function for
```
    virtual void visit(AstNodeCase* nodep) override { unsupported_visit(nodep); }
```
which lead to case statements being treated differently. I'm regressing this as a fix and will create a Pull Request with some snazzy new test_regress tests.
    
Note this fix would probably not work for really obscure and probably not LRM legal code, where the n++ or n-- return value was expected to be the case value:
```
case(foo)
n++: /* do something */ ;
n--: /* do something else */ ;
default: ;
endcase
```

drewranck, 2022-03-10:
https://github.com/verilator/verilator/pull/3349

wsnyder, 2022-03-12:
Thanks for making the pull request.

closed 2022-03-12 by wsnyder