verilator/verilator #3347: Some wires don't appear in VCD files

yurivict, 2022-03-10:
I have the internal wire in one Verilog module:
```
wire           internal_wire[A][B][C-1:0];
```

It doesn't appear in the VCD file at all.

gezalore, 2022-03-10:
Any chance you need to increase https://verilator.org/guide/latest/exe_verilator.html?highlight=trace#cmdoption-trace-max-width ?

yurivict, 2022-03-10:
Adding ```--trace-max-width``` with the bit-width of this wire didn't help.

I also think that it should have helped but it didn't.

wsnyder, 2022-03-12:
Look in the generated .cpp files to see if the signal is listed but commented out, in some cases there is a comment to indicate why it wasn't included.  Otherwise you can try "--debug --debugi-V3TraceDecl 9" and look to see if the reason is indicated in the debug messages.

wsnyder, 2022-04-11:
I'm assuming you figured this out, if not please feel free to post and reopen.

closed 2022-04-11 by wsnyder