verilator/verilator #3348: Some parameters can't be overridden with -G{name}={val}

yurivict, 2022-03-10:
I have the file ```params.v``` with the parameter named ```Param_A``` set there.

The Verilog code uses this parameter ok, but an attempt to override it from the command line fails:
```
%Error: Parameters from the command line were not found in the design: Param_A
```

On the other hand, other parameters can be overridden from the command line successfully.

wsnyder, 2022-03-12:
Where are the parameters in the scope hierarchy?  You can only override top-level parameters.  Otherwise if you think this should work we'll likely will need a test case to explain it.

wsnyder, 2022-05-12:
Did you get this figured out?

yurivict, 2022-05-12:
I'd close this for now because now I don't use it.

I'll reopen if I get back to it.

closed 2022-05-12 by yurivict