verilator/verilator #3351: no scope for verilog dpi export

lwshowl, 2022-03-14:
i have written a task in verilog that do $readmemh and export as dpi function
 it does compile and i used the svSetscope() mentioned in the DPI document
after calling it , it still gets a scope error:

Testbench C called 'initmem' but scope wasn't set, perhaps due to dpi import call without 'context', or missing svSetScope. See IEEE 1800-2017 35.5.3.

here is the task:

    export "DPI-C" task initmem;
    
    task initmem(input string file);
        /* verilator lint_off WIDTH */
        if (file != "") begin
            /* verilator lint_off WIDTH */
            $readmemh(file, cells);
            end else begin
            $display("file not exist");
        end
    endtask

this is the main function:
int main(){

    Vtinyrv32 *dut = new Vtinyrv32;

    svSetScope(svGetScopeFromName("TOP.dut"));

    Verilated::traceEverOn(true);
    VerilatedVcdC *m_trace = new VerilatedVcdC;
    dut->trace(m_trace, 5);
    m_trace->open("waveform.vcd");

    Vtinyrv32::initmem(argv[1]);

   <……>
}

wsnyder, 2022-03-14:
>svSetScope(svGetScopeFromName("TOP.dut"));

I suspect svGetScopeFromName("TOP.dut") is returning nullptr.  Add a test for that (I fixed the docs to note that).  Assuming that is the issue you can see what scopes do exist with Verilated::scopesDump().

wsnyder, 2022-03-14:
Closing assuming that is the issue - feel free to ask followups or reopen.