verilator/verilator #3356: Not all internal state accessible through rootp

shareefj, 2022-03-18:
Hi,

I'm picking up some legacy code that uses hierarchical access to internal member state.  I've found the relevant note about the interface changes in v4.210 and have added `->rootp` liberally throughout the code, but there are some paths that don't seem to be accessible and I don't understand why there are differences between the modules.

As an example, I have a register that was previously accessed via a path such as:

```
top->rootp->top__DOT__a__DOT__b__DOT__c__DOT__d__DOT__e
```

but what I can see now is that in `top___024root.h` I have a number of sub classes listed in the form:

```
VL_MODULE(Vtop___024root) {
    public:
        // CELLS
        Vtop_c* __PVT__top__DOT__a__DOT__b__DOT__c;
```

Looking at all the sub classes, they all seem to be for modules that are instanced more than once.  Is that the difference?  Did this change in v4.x as well?

wsnyder, 2022-03-22:
That there's a __PVT in the name means Verilator thinks the relevant structure doesn't need to be accessible from "outside".  Are you sure the signals you are accessing are marked public appropriately?  If so it's certainly possible a bug got introduced, I would suggest to run with --debug and look at the .tree files to see what signals are marked public.  And/or if you can make an example we can look into it or suggest what you should look into.

shareefj, 2022-03-22:
It doesn't look like we're marking anything as public.  There are no Verilator specific comments in the RTL and I've double checked the old Makefile and it doesn't use any of the `--public` flags.

What am I looking for in the `.tree` files?  A `grep -i public *.tree` doesn't find anything.

wsnyder, 2022-03-22:
In tree files it puts "[PUBLIC]" on tasks/typedefs, and "[P]" on signals/modules.

shareefj, 2022-03-28:
The only public signals that match that description are the top level ports.  We don't declare anything else as public.

I did just try to declare a signal as public to make sure I could see a change but then all hell breaks loose as all the `__DOT__` paths change names and I can't compile.  Looks like in my toplevel .h file I get an additional class so that not everything is accessible via `rootp`.  I think I'm just going to have to bite the bullet and rewrite all the TB code to use DPI.

closed 2022-03-28 by shareefj