verilator/verilator #3357: Generating Module Instance using Verilator

shardator, 2022-03-19:
How may we help - what is your question?
Hi, I'm curious if there is a way to use Verilator to not just lint or compile a Verilog file, but also generate an instantiation. For large modules creating instantiation is a tedious and it would be good to just write for `something.v`

```
module something(input x, output y);
...
endmodule
```

the command:

```
verilator --dump-instantiation something.v
```

would magically print

```
something(  .x( _in_x  ),
            .y( _out_y ) );
```

or such, even including parameters, etc. Obviously this is more useful for large modules.

wsnyder, 2022-03-19:
Verilator doesn't have such an option, though it would be easy for you to make an extension using the --xml option.

However I think what you likely want instead the batch mode that is part of [verilog-mode](https://veripool.org/verilog-mode/) and specifically "/*AUTOINST*/"

closed 2022-03-19 by wsnyder

shardator, 2022-03-19:
Well, `--xml-only` is almost there, the only problem is that it requires the specification of all the params, and it would be great if the params would be part of the output. So this could be an extension to more like the `--lint` option, which does not require params.

Well, verilog-mode is nice, but it is for emacs. It is suboptimal to include emacs in a workflow just for this.

wsnyder, 2022-03-19:
Verilator by its nature must elaborate, so there's no way it will ever support keeping the parameters "opaque".

View Emacs in the Verilog-Mode context as a programming language, yes you need to install it (if not already installed which it usually is) but otherwise it's no different from Python or any other language you should be able to ignore underneath the tool.

gezalore, 2022-03-19:
`emacs --batch -Q -l $VERILOG_MODE_EL $INPUT_FILE -f verilog-batch-auto`

shardator, 2022-03-19:
> Verilator by its nature must elaborate, so there's no way it will ever support keeping the parameters "opaque".
> 

Wishful thinking: translate parameters into C++ template arguments. (It would not be impossible in theory.)

> View Emacs in the Verilog-Mode context as a programming language, yes you need to install it (if not already installed which it usually is) but otherwise it's no different from Python or any other language you should be able to ignore underneath the tool.

Yes, never used emacs, but with the comment from @gezalore I understand it. I think I will try to write my own parser for this though, having emacs installed is probably a bit heavy weight solution.

wsnyder, 2022-03-19:
Wouldn't discourage writing a tool... but still would suggest give the AUTOs a quick try.  They are extremely widely used - if you have any non-trivial IP from third parties most likely you'll find some AUTO in them & a quick github search shows over 51,000 AUTOINST code usages.