verilator/verilator #3359: Logic error on V3Partition

HungMingWu, 2022-03-22:
In V3Partition.cpp
``` cpp
class PartFixDataHazards final {
    // Ignored
private:
    void findAdjacentTasks(OvvSet::iterator ovvIt, TasksByRank* tasksByRankp) {
        // Ignored
        for (V3GraphEdge* edgep = (*ovvIt)->outBeginp(); edgep; edgep = edgep->outNextp()) {
            const OrderLogicVertex* const logicp = dynamic_cast<OrderLogicVertex*>(edgep->fromp());
            // Ignored
        }
    }
}
```
I think this is a logic error, enumerate out edges should use top, not fromp.

wsnyder, 2022-04-13:
@gezalore does this look right?

gezalore, 2022-04-13:
The issue seems genuine. `fromp()` of an out edge should be the same as the vertex (in this case `*ovvIt`). Worse yet, `*ovvIt` is always an `OrderVarStdVertex`, so the dynamic cast always yields `nullptr`, the rest of the loop is then dead. You can indeed see this in the code coverage report: https://app.codecov.io/gh/verilator/verilator/blob/master/src/V3Partition.cpp (Line 1852). 

I will run a few benchmarks with this change. It does perturb some of the partitioning quite a lot (e.g.: t_bench_mux4k goes from having 43 mtaks to 2 mtasks, which is a bit concerning).

gezalore, 2022-04-13:
BTW: for a while now I wanted to Add `V3GraphVertex::foreachInEdge`, `V3GraphVertex::foreachOutEdge`, `V3GraphVertex::foreachInVertex`, and , `V3GraphVertex::foreachOutVertex`. Feeling more motivated now..

wsnyder, 2022-04-14:
@jcoiner?

gezalore, 2022-08-26:
I have done some analysis of this and although the reported bug is present in the code, it is not actually a serious issue.

The purpose of this pass (PartFixDataHazards) is to enforce an ordering between otherwise un-ordered reads and writes of the same variable (as those might execute on different threads).

write-write hazards are problematic because some writes in SV are actually Read-modify-write in C++ as explained at the top of the class comment, but the basic examples is:
```systemverilog
reg [1:0] a;
always @(posedge clk) a[0] = x;
always @(posedge clk) a[1] = y;
```
These are unordered but the first turns into `a = (a & 2) | x`, and he second is `a = (y << 1) | (a & 1)`, so these need to be serialised. 

The current code correctly fixes these write-write hazards.

unordered write-read hazards can also exist:
```systemverilog
module top(
  input wire clk,
  output reg a,
  output reg b,
  input  reg c
);

  always @(posedge clk) a = b;
  always @(posedge clk) b = c;

endmodule
```

Here the result is different based on the execution order of the two otherwise un-ordered assignments. The current code does not fix this (and when verilating the above with `--threads 2`, they end up on two different threads, so the simulation is nondeterministic (yet still in line with the scheduling semantics so long as the update of `b` is atomic).

More interestingly, even with fixing the reported bug, the current code still does not fix this, because  it only considers OrderVarStdVertex variable vertices in the input dependency graph, which do not actually represent this hazard at all: the read of `b` is in a clocked process, and clocked processes do not have consumed variables as OrderVarStdVertex dependencies in the order/move graph, only combinational logic have those.

In general the OrderVarStdVertex variable dependencies always starts from a clocked logic block, and is chained through combinational logic blocks, e.g.: `clocked->var1->comb1->var2->comb2`, i.e.: the OrderVarStdVertex dependency graph considered here does not represent variables read by clocked processes.

Thinking in the context of v5, all the other cases of read-write hazards mentioned at the top of the algorithm in the comment are not applicable (combinational loops are broken and detected externally, and together with the triggers being computed outside of the ordered code they do not cause a problem, and the clockenable thingum does not exist anymore).

In short I think:
1. The current code correctly fixes write-write hazards (these are the most dangerous)
2. There are no read-write hazards through combinational logic (those are covered by the dependency graph built by V3Order)
3. Does not fix other read-write hazards, but these arise only due to race conditions in clocked input code, not due to anything done by Verilator itself, and the SV standard does not specify the ordering of these.

Point 3 is the tricky one. SV allows for the statements of multiple always blocks to be inter-leaved (IEEE-1800 4.6/4.7) so it's not an obvious problem, but a single statement (or at least a single simple statement? the standard is unclear here) in an always block still needs to execute atomically. As the example above, I think wide variable assignments can violate this:
```
  always @(posedge clk) a = b;
  always @(posedge clk) b = c;
```
Note that this is racy anyway, but according to SV I think you can only have two outcomes: all three variables are equal, or `a` is the old value of `b`). With Verilator, if variables `a`, `b`, and `c` are wider than a machine word that can be atomically transferred, then `a` can become half old value of `b` and half `c` if you execute these on different threads, which is not valid in SV.

In short I think the current code is consistent with SV scheduling, except with some input that is intrinsically racy anyway, in which case you might end up with a non-standard compliant evaluation, but in that case a standard compliant evaluation would be non-deterministic anyway.

Does that mean that we can just ignore read-write hazards at all (like we effectively do now, due to the reported bug), so long as we are happy to live with a more non-deterministic simulation than alowed by the standards, or should we force ordering of these blocks?

Regarding the slowdown, it is caused by all reads of a variable being serialised. And this comment in particular being inaccurate regarding the cost of doing so:
https://github.com/verilator/verilator/blob/5869fdf7f6f43103c5a7d09b9d0a4f21774bcb29/src/V3Partition.cpp#L2116-L2130

wsnyder, 2025-09-26:
98206a4f04 last year cleaned up the offending code.

gezalore, 2026-08-18:
Fixed by #8133