verilator/verilator #3362: Verilator - memory usage during compilation

agrobman, 2022-03-23:
I was compiling 8 cores cluster design - the process allocated ~139GB memory. the simulation of the same design allocated ~70MB memory. 
Is there any way to reduce memory demand during compilation ? ( we use GCC flow and it's 4.210 version of verilator.)

Another issue. While xrun simulates faster if a RTL design isn't clocked , seems verilator is not. Is there possible simulation speed optimization opportunity?  (no clock do not evaluate a module)...

Our design has a lot of microarchitectural clock gating ...

wsnyder, 2022-03-26:
I'd prefer if you didn't mix issues in a ticket, as different answers/followups might be needed, thanks.

If using --threaded we made some memory fixes in the trunk version, but likely it won't be massively better.

See if the clock gaters are properly recognized as such with the /*verilator clock_gater*/ alternatively make sure RV_FPGA_OPTIMIZE (assuming this looks like SwERV designs) is set, as this helps Verilator recognize the gating.

agrobman, 2022-03-28:
I was thinking creation and than using a library (archive) may reduce memory demand during linking in expense of compilation time (?).

where should I place the clock_gater comment ?  near the clock gating cell module? enable signal?
Now it looks like:

```
module CGC (
   input logic SE, EN, CK,
   output Q
   );
    logic  en_ff /*verilator clock_enable*/;
    logic  enable;
  
   assign      enable = EN | SE;
 
   always_latch begin
      if(!CK)
        en_ff = enable;
   end

   assign Q = CK & en_ff;
 
endmodule 
```

wsnyder, 2022-12-16:
Think this was answered, no specific test to fix, so closing.

closed 2022-12-16 by wsnyder