verilator/verilator #3364: void'(some_queue.pop_front()); in the body of an if-stmt lints cleanly but fails to produce good cpp

drewranck, 2022-03-25:
Seems related to a closed issue:
https://github.com/verilator/verilator/issues/1641

I will attempt to fix this myself, and I have to create a test_regress for it, but the gist is:

```
always @(posedge clk) begin : some_queue_handler_fails
  if (some_queue.size()) begin
    void'(some_queue.pop_front());
  end
end

always @(posedge clk) begin : other_queue_handler_good
  if (other_queue.size()) begin
    other_queue = other_queue[1:$]; // pop front explicitly
  end
end
```

The 'some_queue' case is giving me a verilator error like:
```Vblahblah__DepSet_h311ab79c__0.cpp: In function ‘void Vblahblah___sequent__TOP__blah__0(Vblahblah*)’:
Vblahblah__DepSet_h311ab79c__0.cpp:99:53: error: expected ‘;’ before ‘}’ token
             vlSelf->__PVT__some_queue.pop_front()}
                                                     ^
                                                     ;
make[1]: *** [/foo/verilated.mk:257: Vblah__ALL.o] Error 1
```

>What 'verilator --version' are you using?  Did you try it with the git master version?

Verilator 4.219 devel rev v4.218-44-gb5791192

> What OS and distribution are you using?

Debian, seems unrelated to OS though

> May we assist you in trying to fix this in Verilator yourself?

I will try to track it down, seeing as explicit queue manipulation works.

drewranck, 2022-03-25:
I feel dirty using a temp macro workaround in the meantime.

```
`ifndef void_pop_front
`define void_pop_front(queue)                                       \
  `ifdef VERILATOR                                                  \
    if (queue.size() <= 1) queue.delete(); else queue = queue[1:$]; \
  `else                                                             \
    void'(queue.pop_front());                                       \
  `endif
`endif
```

wsnyder, 2022-03-25:
Most likely the issue is V3Width isn't properly calling ->makeStatement(). That's what adds the needed semicolon.  If you can look into it probably won't take you much time to track it down, thanks!

drewranck, 2022-03-28:
I'll admit, I'm still not an expert on the Verilator Ast methods.

I ran a simple testcase with: --debug --dump-treei 9.

I noticed that if I have something like `if (queue.size() > 0) item = queue.pop_front();`, I get a tree like this, with an ASSIGN:
```
    1:2:3:2:2: IF 0x555556a8e180 <e239> {d34al}
    1:2:3:2:2:1: GT 0x5555569fad10 <e240> {d34bj} @dt=0x555556a940d0@(G/w1)
...
    1:2:3:2:2:2: ASSIGN 0x5555569fae70 <e236> {d34bt} @dt=0@
    1:2:3:2:2:2:1: METHODCALL 0x555556a9c640 <e567> {d34ci} @dt=0@  pop_front ->  -> UNLINKED
```

If I have something like `if (queue.size() > 0) void'(queue.pop_front());` then the parent tree node of the METHODCALL does not have an assign:

```
    1:2:3:2:2: IF 0x555556a8e300 <e315> {d38al}
    1:2:3:2:2:1: GT 0x5555569fb3f0 <e316> {d38bj} @dt=0x555556a940d0@(G/w1)
    ...
    1:2:3:2:2:2: METHODCALL 0x555556a9cb40 <e617> {d38ch} @dt=0@  pop_front ->  -> UNLINKED
```    

I tried a few things unsuccessfully in V3Width.cpp for the makeStatement() call, but I wasn't sure how to use V3Ast methods to check if the parent was an ASSIGN.

Maybe instead of what's currently there for pop_front: `if (!nodep->firstAbovep()) newp->makeStatement();`

Something like "if we're not the RHS of something, then add a ;":
```
	     if (!(nodep->abovep && 
	           nodep->abovep->rhsp &&
		   nodep->abovep->rhsp  == nodep)) newp->makeStatement();
```

wsnyder, 2022-03-28:
Looking at other code there, try

       if (!nodep->firstAbovep()) newp->makeStatement();

drewranck, 2022-03-29:
V3Width.cpp for queue cases "pop_back" or "pop_front" already has that line:

```
	    if (!nodep->firstAbovep()) newp->makeStatement();
```

I think the problem is more subtle, because firstAbovep() doesn't tell us if what type of SV code the parent node is. I was kind of hoping something like nodep->abovep could tell us if we were in a "block" or something.

I added some printf debugging in the V3Width.cpp to help illustrate this:

```
        } else if (nodep->name() == "pop_front" || nodep->name() == "pop_back") {
            methodOkArguments(nodep, 0, 0);
            // Returns element, so method both consumes (reads) and modifies the queue
            methodCallLValueRecurse(nodep, nodep->fromp(), VAccess::READWRITE);
            newp = new AstCMethodHard(nodep->fileline(), nodep->fromp()->unlinkFrBack(),
                                      nodep->name(), nullptr);
            newp->dtypeFrom(adtypep->subDTypep());
	    if (!nodep->firstAbovep()) newp->makeStatement();
	    // TODO(drewranck): hacking in progress:
	    UINFO(1, "queue op, " << nodep << endl);
	    UINFO(1, "   abovep(): " << nodep->abovep() << endl);
	    UINFO(1, "   firstAbovep(): " << nodep->firstAbovep() << endl);
```

This gives some interesting results for various places when using void'(queue.pop_front()). 

For example, if there is any dummy line of code in a block before the void'(..) call, the ; is correctly added:

```
      if (my_int_queue.size() > 0) begin
        $display("hello another dummy line");
        // V3Width debug -- firstAbovep()=nullptr (good)
        void'(my_int_queue.pop_back());
      end
```

If there is no line of code before the void'(..) call, the ; is not added:

```
      if (my_int_queue.size() > 0) begin
        // V3Width debug -- firstAbovep()=BEGIN (bad, we get no ;)
        void'(my_int_queue.pop_back());
      end
```

I could write a helper function in V3Ast.h, instead of firstAbovep(), something that tells us if the parent is "block" or "body" of something (aka, BEGIN, IF, CASEITEM, etc, I'm not sure all the V3Ast Node types) then we need the ; added, but if it's some other statement/conditional (LT, GTE, ASSIGN, etc) then we don't need the ; added.

wsnyder, 2022-03-31:
I think isStandaloneStmt() is likely to be error prone, as I'm not sure what you have is right, and adding new node types might forget to add this.  Perhaps instead on visiting the Cast that is void, change the void's right hand side into a statement?

drewranck, 2022-04-01:
I agree that what I have could be error prone, if some new Node type didn't use one of the ones I have as a parent class (b/c `VN_IS(..)` does a > compare, so it catches all the existing ones and their children). Although `firstAbovep()` is already error prone.

There is no "void" node, right? I don't know that there needs to be one either. Modelsim will only give a warning if you don't add an explicit void'(..) cast on a function with return value. 

I guess this comes back to me not knowing all the details about V3Ast nodes. As an example, something like an IF node looks like:

IF
  -> condp() Condition (such as GT)
  -> ifsp() if-true list
  -> elsep() else-list
  
So ifsp()'s list has 3 statements s0, s1, s2; if s0 happens to be any node that relies on firstAbovep() working, it may not generate cpp that can compile.

One bad thought I had was, and this might hurt everyone's Ast node/tree performance, is that every "list of statements" chain would start with some additional new Nop node, so the previous 3 statements list would now look like Nop, s0, s1, s2 -- and now statement s0 would work correctly with firstAbovep() (s0 is no longer first in the list).

Going back to the commit I have, and looking at the base AstNode class, I guess I'm struggling with how else to know I'm the first statement in a list of statements. Like, to correctly check when the backp() is an AstNodeIf, I'd need to check something like:
```
backp() && 
VN_IS(backp(), NodeIf) && 
(VN_AS(backp(), NodeIf).ifsp() == nodep() ||
 VN_AS(backp(), NodeIf).elsesp() == nodep())
 ```
 which is kind of a lot of checking, and again error prone. What if instead, class `AstNode` had a virtual method like: `bool isFirstInOneOfMyListOfStatements(AstNode* n) const { return false; }`, and then class `AstNodeIf` should implement the method as:
`bool isFirstInOneOfMyListOfStatements(AstNode* n) const { return (n == ifsp() || n == elsep()); }`
and then `isStandaloneStmt()` would be less error prone:
```
return (!firstAbovep() || /* 2nd or later in list */
        (backp() && backp().isFirstInOneOfMyListOfStatements(this))); /* first in list */
```

drewranck, 2022-04-01:
Switch branched on my fork:
https://github.com/drewranck/verilator/tree/issue-3364-try2

drewranck, 2022-08-11:
https://github.com/drewranck/verilator/tree/issue-3364-try3

wsnyder, 2022-08-12:
I appreciate your pull request to fix this!

closed 2022-08-12 by wsnyder