verilator/verilator #3366: How to fix the compile errors in msys [WINDOWS 10]

elike-ypq, 2022-03-30:
`making verilator in src
make -C src
make[1]: Entering directory '/home/elike/verilator/src'
make -C obj_dbg -j 1  TGT=../../bin/verilator_bin_dbg.exe VL_DEBUG=1 -f ../Makefile_obj serial
make[2]: Entering directory '/home/elike/verilator/src/obj_dbg'
If you get errors from verilog.y below, try upgrading bison to version 1.875 or newer.
/mingw64/bin/python3 ../bisonpre --yacc /usr/bin/bison -d -v -o V3ParseBison.c ../verilog.y
  edit ../verilog.y V3ParseBison_pretmp.y
  C:/msys64/usr/bin/bison -d -v --report=itemset --report=lookahead -b V3ParseBison_pretmp -o V3ParseBison_pretmp.c V3ParseBison_pretmp.y
  edit V3ParseBison_pretmp.output V3ParseBison.output
Traceback (most recent call last):
  File "C:\msys64\home\elike\verilator\src\bisonpre", line 552, in <module>
    process()
  File "C:\msys64\home\elike\verilator\src\bisonpre", line 42, in process
    clean_output(tmp_prefix() + ".output",
  File "C:\msys64\home\elike\verilator\src\bisonpre", line 105, in clean_output
    lines = fh.readlines()
UnicodeDecodeError: 'gbk' codec can't decode byte 0xa2 in position 7234: illegal multibyte sequence
make[2]: *** [../Makefile_obj:343: V3ParseBison.c] Error 1
make[2]: Leaving directory '/home/elike/verilator/src/obj_dbg'
make[1]: *** [Makefile:65: ../bin/verilator_bin_dbg.exe] Error 2
make[1]: Leaving directory '/home/elike/verilator/src'
make: *** [Makefile:141: verilator_exe] Error 2`

elike-ypq, 2022-03-31:
> bison --version
bison (GNU Bison) 3.8.2
Written by Robert Corbett and Richard Stallman.

Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
[V3ParseBison_pretmp.zip](https://github.com/verilator/verilator/files/8385959/V3ParseBison_pretmp.zip)

wsnyder, 2022-03-31:
Please try the git master version, the latest bison generates Unicode

wsnyder, 2022-04-11:
Assuming it's working, if not please reopen.

closed 2022-04-11 by wsnyder

urischachar, 2022-06-11:
problem still occured with bison 3.8.2
There is a Epsilon character.
I'm following "Installing Verilator on Windows 10 using a fresh MSYS2 install"

Traceback (most recent call last):
  File "C:\msys64\home\user\verilator\src\bisonpre", line 560, in <module>
    process()
  File "C:\msys64\home\user\verilator\src\bisonpre", line 42, in process
    clean_output(tmp_prefix() + ".output",
  File "C:\msys64\home\user\verilator\src\bisonpre", line 166, in clean_output
    print(line)
  File "C:/msys64/mingw32/lib/python3.9/encodings/cp1255.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u03b5' in position 19: character maps to <undefined>

wsnyder, 2022-06-11:
>File "C:\msys64\home\user\verilator\src\bisonpre", line 166, in clean_output
print(line)

You're not using the most recent version, as there's no print at that line number, so I suspect you don't have the code with this fix in it.

urischachar, 2022-06-11:
I just added some printing to understand what happen.
It seems that line 164 is trying to print an Epsilon sign ('\u03b5') which exist in the **V3ParseBison_pretmp.output** file.
Here is the original error:

$ /mingw32/bin/python3 ../bisonpre --yacc /usr/bin/bison -d -v -o V3ParseBison.c ../verilog.y
  edit ../verilog.y V3ParseBison_pretmp.y
  C:/msys64/usr/bin/bison -d -v --report=itemset --report=lookahead -b V3ParseBison_pretmp -o V3ParseBison_pretmp.c V3ParseBison_pretmp.y
  edit V3ParseBison_pretmp.output V3ParseBison.output
Traceback (most recent call last):
  File "C:\msys64\home\user\verilator\src\bisonpre", line 536, in <module>
    process()
  File "C:\msys64\home\user\verilator\src\bisonpre", line 42, in process
    clean_output(tmp_prefix() + ".output",
  File "C:\msys64\home\user\verilator\src\bisonpre", line 164, in clean_output
    fh.write(line)
  File "C:/msys64/mingw32/lib/python3.9/encodings/cp1255.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u03b5' in position 19: character maps to <undefined>