verilator/verilator #3367: Cannot compile osu018_stdcells.v: could we add support for table primitives please?

hughperkins, 2022-04-03:
Cannot compile osu018_stdcells.v, error messages:

```
$ verilator --sv --cc tech/osu018/osu018_stdcells.v
%Error: tech/osu018/osu018_stdcells.v:241:12: syntax error, unexpected '(', expecting IDENTIFIER or randomize
  241 |    udp_dff (DS0000, D, I0_CLOCK, 1'B0, 1'B0, NOTIFIER);
      |            ^
%Error: tech/osu018/osu018_stdcells.v:279:12: syntax error, unexpected '(', expecting IDENTIFIER or randomize
  279 |    udp_dff (DS0000, D, CLK, 1'B0, 1'B0, NOTIFIER);
      |            ^
%Error: tech/osu018/osu018_stdcells.v:321:12: syntax error, unexpected '(', expecting IDENTIFIER or randomize
  321 |    udp_dff (P0003, D_, CLK, I0_SET, I0_CLEAR, NOTIFIER);
      |            ^
%Error: tech/osu018/osu018_stdcells.v:552:13: syntax error, unexpected '(', expecting IDENTIFIER or randomize
  552 |    udp_tlat (DS0000, D, CLK, 1'B0, 1'B0, NOTIFIER);
      |             ^
%Error: tech/osu018/osu018_stdcells.v:591:13: syntax error, unexpected '(', expecting IDENTIFIER or randomize
  591 |    udp_mux2 (I0_out, B, A, S);
      |             ^
%Error: Exiting due to 5 error(s)

```

To reproduce:
- download osu018_stdcells.v from https://download.libresilicon.com/technologies/OSU018/
- `verilator --sv --cc osu018_stdcells.v`

Tested with version 4.218

Looking through https://verilator.org/guide/latest/languages.html , looks like the issue is that primitive tables are not currently supported. Dont suppose... could it be possible to add support for these, in order to enable gate-level simulation please? (verilator should be great for gate-level simulation, since verilator runs so fast :) ).

wsnyder, 2022-04-03:
Please see  #468.  That no one has contributed this yet indicates it continues to not be a high priority in the middle-term, so perhaps you would be willing to work on it?  If so please post on #468.

closed 2022-04-03 by wsnyder

hughperkins, 2022-04-03:
Ok, understood. What are standard approach(es) for gate-level simulation, using verilator? Is there e.g. a version of osu018_stdcells.v that uses a different approach instead of anonymous user designed primitives?

wsnyder, 2022-04-03:
I don't know about any specific libraries, ideally whoever provides it would also provide a RTL version of each cell, as some other vendors do.  Alternatively you'd need to hand make that, then ideally prove equivalence with a formal solver.  Although libraries are huge, there's generally not that many unique cell types, as the same RTL function is often replicated across different Vt strengths, etc.

hughperkins, 2022-04-03:
There's a bunch of available files, https://download.libresilicon.com/technologies/OSU018/ , but none ending in `.rtl`. rtl file is verilog, and verilator can handle it? What would an RTL file use to represent a flip-flop cell?

wsnyder, 2022-04-03:
The file would end in .v, and have an "always @ (posedge clk)" or something.  I suspect there's a libresilicon forum that might have advice.

hughperkins, 2022-04-03:
(Do you happen to know any other freely available cell library that has the required form? I'm not wedded to osu018, it's just what I found to hand....; I'm just using it to get some kind of idea of relative propagation delay, area, etc (using iverilog, tons of stuff failed on gate-level simulation, because of the whole `x` thing changing behavior of `if` statements in behavioral vs gate-level version of the `if` statement; it's possible that such simulation is not required for verilator, but would prefer to try it, just in case) )

wsnyder, 2022-04-03:
I don't, myself.

hughperkins, 2022-04-03:
Ok, thanks! :)

hughperkins, 2022-04-03:
In case anyone else comes across this issue, I just ended up modifying osu018_stdcells.v, in this change, https://github.com/hughperkins/VeriGPU/commit/fc16016291967f3a4de99247a81378c0788a4699 . I can then compile this with verilator, and stuff runs ok. Not sure if I'm testing anything useful using this, or whether my changes are correct (I threw away the NOTIFIER for now; which is probably a bad idea, since I suppose it's used to model delay propagation?... but stuff didn't run when I tried to include NOTIFIER).

hughperkins, 2022-04-03:
Ok, so, if I put in some hacky NOTIFIER code, but make sure `rst` in my own circuit lasts an extra clock cycle, then stuff runs ok for me, and I get rid of the icky feeling of 'cheating' on the GLS. https://github.com/hughperkins/VeriGPU/commit/152b66044f4f907622f79fedd337d5e80398d6a7

mithro, 2022-04-03:
FYI - You might want to consider the fully open source, manufacturable SkyWater 130nm PDK found at https://github.com/google/skywater-pdk -- When installed with https://github.com/RTimothyEdwards/open_pdks - I believe it can be used for gate-level simulation?