verilator/verilator #3369: How to get the number of flops/registers in the design?

yurivict, 2022-04-08:
This is a valuable metric that can help to evaluate the size of the design, the number of transistors/gates.

Does Verilator print this information?

margej, 2022-04-08:
Maybe Verilator does this now, I've been out of the llop for about a year.
If not, I wrote a patch to do exactly this as part of V3Stats. I'll see if I can find it...

margej, 2022-04-08:
I found the branch and it looks like it still applies cleanly to master, so let me know if it's of interest and I can try and get a PR out

yurivict, 2022-04-08:
> [...] let me know if it's of interest and I can try and get a PR out

Thank you, it is definitely of interest to me.

wsnyder, 2022-04-09:
@margej saw you made #3372 then closed it, I assume it just wasn't ready yet, if so that's fine, just want to make sure nothing was pending to be reviewed.

As to the proposed patch, I'd prefer it live as part of the existing visitor if it can, rather then yet another iteration (which are slow).

margej, 2022-04-11:
Yes, sorry, I thought I was only raising a draft PR. I'm a total novice with the github interface!

I can try and add my new node visitors to the existing StatsVisitor class instead of creating the new StorageCountVisitor. Is that what you are suggesting?

wsnyder, 2022-04-11:
>add my new node visitors to the existing StatsVisitor class instead of creating the new StorageCountVisitor. Is that what you are suggesting?

Yes, thanks.

wsnyder, 2022-07-08:
Are you interested in finishing this off, or is it no longer needed?

yurivict, 2022-07-08:
This is a useful feature and it is generally needed.

margej, 2022-07-09:
Sorry, I'd forgotten it was still open to be honest. I would like to get this cleaned up don't have time at the moment. I'll see if I can have another look at getting it in soon ...

wsnyder, 2022-10-21:
Closing - we'll take a pull to add this, but doesn't seem to be something we must remember to implement.

closed 2022-10-21 by wsnyder